mne-tools/mne-python #5668: [FEATURE] Allow reading EEGLAB standalone .set files with `preload=False`

sappelhoff, 2018-10-27:
EEGLAB `.set` files are essentially MATLAB `.mat` files with a struct datatype `EEG` that contains data fields. One of these fields is `data` and it can include a path to an associated `.fdt` ("float data") file with the binary data.

However, it can also include the data directly. This is then called a "standalone `.set`" file. For these files, MNE-Python's `preload=False` does not work yet.

We should look into making `preload=True` an option by trying to read only selective fields of the `.set` file into memory.

A starting point is [here](https://de.mathworks.com/help/matlab/import_export/load-parts-of-variables-from-mat-files.html)

This would benefit MNE-BIDS, see [discussion](https://github.com/mne-tools/mne-bids/pull/106#issuecomment-433591259)

sappelhoff, 2018-10-27:
yes, or in fact [`pymatreader`](https://gitlab.com/obob/pymatreader)?

https://github.com/mne-tools/mne-python/blob/46585fb139a15d024941eb24273f7a0ab097515c/mne/io/eeglab/eeglab.py#L42

sappelhoff, 2018-10-27:
more information on this topic: It might be impossible to implement, because *fields* (`data`) of *structs* (`EEG`) in *.mat* files (`.set`) cannot be accessed:

https://de.mathworks.com/matlabcentral/answers/339269-how-to-get-a-specific-field-from-a-structure-stored-in-mat-file#comment_452043

... unless these matfiles are saved using a `-struct` option to begin with: see [here](https://de.mathworks.com/matlabcentral/answers/98870-is-it-possible-to-use-the-load-function-to-load-structure-fields-selectively-in-matlab#answer_108218)

Now we need to find out how EEGLAB saves the `.set` files. But I don't think they use the `-struct` option, so it's looking bad for this issue.

jasmainak, 2018-10-27:
EEGLAB has two ways of saving files? One as two files and the other as a single file. Do you know which format they recommend? Maybe for `mne-bids` we can simply ask users to re-save their files in the multi-file format?

jasmainak, 2018-10-27:
For the two file format, it's already implemented :)

agramfort, 2018-10-28:
anything else we can do then?

sappelhoff, 2018-10-28:
>  IIRC the two-file system is the default.

I think the default is saving as a standalone `.set` file, see [here](http://cognitrn.psych.indiana.edu/busey/temp/eeglabtutorial4.301/allfunctions/eeg_options.m)

> Maybe for mne-bids we can simply ask users to re-save their files in the multi-file format?

@jasmainak - yes, I think this is what we should do. We should have a warning or some kind of log message for this.

> anything else we can do then?

nothing to do from MNE-Python's side, we'll have to deal with it in MNE-BIDS. Thanks for the discussion!

closed 2018-10-28 by sappelhoff

mmagnuski, 2018-10-28:
@jasmainak Oh, cool, I didn't know - I always preload. :)