mne-tools/mne-python #5670: inconsistencies in edf reader and annotation readers

jasmainak, 2018-10-27:
I have a couple of issues here but bunching them together as they seem related.

### 1. EDF reader should work with preload=False

If I do:

```py
import mne
from mne.datasets import eegbci
edf_path = eegbci.load_data(subject=1, runs=2)[0]
raw = mne.io.read_raw_edf(edf_path, preload=False)
```

I get

```py
RuntimeError: EDF+ Annotations (TAL) channel needs to be parsed completely on loading. You must set preload parameter to True.
```

However, I can actually do:
```py
import mne
from mne.datasets import eegbci
from mne.io.edf.edf import read_annotations_edf

edf_path = eegbci.load_data(subject=1, runs=2)[0]
raw = mne.io.read_raw_edf(edf_path, preload=False, stim_channel=None)

annot = read_annotations_edf(edf_path)
raw.set_annotations(annot)
```

and it gives me the same result as:
```py
raw2 = mne.io.read_raw_edf(edf_path, preload=True)
assert raw.annotations.duration[0] == raw2.annotations.duration[0]
```

So, it seems to me that it should already be possible to load EDF files with `preload=False` as all the necessary components already exist.

### 2. find_edf_events and mne.events_from_annotations don't give same results

Here is the code to reproduce (continuing from code above):

```py
events, event_id = mne.events_from_annotations(raw)
events2 = raw.find_edf_events()
print(events)
print(events2)
```
### 3. namespaces of read_annotations_xxx

I have to do:

```py
from mne import read_annotations  # for fif files
from mne.io import read_annotations_brainvision  # for brainvision
from mne.io import read_annotations_eeglab  # for eeglab
from mne.io.edf.edf import read_annotations_edf  # for edf
```

This is quite confusing. In fact, I don't see a good rational for having different functions as all these different file formats (debatable) since all you need is a filename for most of these readers. So you should be able to triage based on the extension. In any case, having them in the same namespace would already be a start ! (and maybe `read_annotations` -> `read_annotations_fif` ?)

### 4. documentation issues

`read_annotations_edf` says I [need a `.vmrk` file](https://github.com/mne-tools/mne-python/blob/master/mne/io/edf/edf.py#L1127). This looks incorrect, right?

Happy to help out with these after the `mne-bids` release. If someone else has time to work before (@massich ?), I am happy to review the code :)

agramfort, 2018-10-28:
thanks heaps @jasmainak for the details report. Yes @massich should be the first person to start addressing these issues. 

Just regarding 2: `raw.find_edf_events()` should be deprecated unless it gives you something you cannot do with events_from_annotations

regarding 4: 
yes it should be `from mne.io import read_annotations_edf` and I agree that read_annotations should / could triage based on file extension, especially since we'll soon have IO for CSV files.

cbrnr, 2018-10-28:
These are all excellent points. @massich I'd be happy to help address these issues. Regarding 3, I've brought this up once for the loading functions (`read_raw_XXX`), but most people didn't like the idea. I still do, having one function to load the data and one function to read annotations seems much simpler, given that we can determine the file type by its extensions (or if that doesn't work we can still parse the magic bytes).

cbrnr, 2018-10-29:
Re 2, actually `raw.find_edf_events` gives you back the annotations as they are stored in the EDF file (no stim chan is synthesized). It is deprecated because all the information is contained in `raw.annotations`. I don't know the plan about what is going to happen with events - will `mne.find_events` still be an option? Or will it be replaced by `mne.events_from_annotations`? In this example, they contain the same events, only in a different format. Is that what you meant by different?

massich, 2018-10-30:
#### re 1

Yes, that's expected. What it could be good is to set `stim_channel=None` as default. And warn. But I'm not sure how this plays in **mne-python**'s timeline for deprecationg stim channel.

The other thing, is that you cannot read the TAL channel in the edf_file on `preload=False` unless you parse the entire file. So I'm not sure. If you supply a file with only annotations, then it would make sense.

#### re 2

this is weird, I thought we had a test for that. 

#### re 3
I also dislike this `_XXXX` in the readers. But for now I would not touch it.

#### re 4
my bad.

jasmainak, 2018-10-30:
> The other thing, is that you cannot read the TAL channel in the edf_file on preload=False unless you parse the entire file. So I'm not sure. If you supply a file with only annotations, then it would make sense.

If you know which memory segments to access for the TAL channel, wouldn't this be fast? How fast is `read_annotations_edf` for really long recordings like say, sleep data?

massich, 2018-10-30:
>If you know which memory segments to access for the TAL channel, wouldn't this be fast? How fast is read_annotations_edf for really long recordings like say, sleep data?

yes it should be plenty doable. The only thing is that this was not present in the EDF reader, and we did not implement it.

agramfort, 2018-10-30:
agreed regarding TAL channel parsing without preload it should be doable but it's presently not implemented. To be done at some point. No rush on my side.

Regarding 4 as discussed with @massich I would suggest to:

 * make private : read_annotations_brainvision read_annotations_eeglab read_annotations_edf 

 * add triage to the private functions based on file extension for mne.read_annotations

larsoner, 2018-11-05:
FYI to me it is weird to have a `read_annotations` that triages by file type but no `read_raw` that does the same. The more consistent fix would be to have `read_annotations` renamed to `read_annotations_fif`. But I can live with the inconsistency.

larsoner, 2018-11-05:
I suspect that after this annot refactoring we should have many fewer parameters

closed 2018-11-06 by massich

massich, 2018-11-06:
Maybe I've merged it too soon. I did not see that the conversation was still alive.

larsoner, 2018-11-06:
The conversation went a bit off-topic (to `mne.io.read_raw`), I think it's fine to leave closed

PR #5695 (2018-11-06): [MRG+1] Unify read annotations
Fixes #5670 

Things to do:
- [x] migrate the calls of:
  - [x] brainstorm
  - [x] brainvision
  - [x] EDF
  - [x] fif
  - [x] eeglab
- [x] Depreacate read_annotations_eeglab since it was in 0.16
```rst - Add function :func:`mne.io.read_annotations_eeglab` to allow loading annotations from EEGLAB files, by `Alex Gramfort`_` ```
commit ae38e1b7 (2018-11-06): [ENH] [MRG+1] Unify read annotations (#5695)

Unify all annotation readers under `mne.read_annotations`.

Fixes #5670