mne-tools/mne-python #5671: private attribute to detect if raw has been cropped?

jasmainak, 2018-10-28:
In mne-bids, we were thinking of an API where there is a function called `write_raw_bids(raw, bids_fname, ...)` to convert existing data to BIDS format. This would work by essentially **copying** the original file in raw.filenames to the new destination and converting the metadata into json and tsv sidecar files (it's the only way we can support different file types since MNE can only write fif files). 

However, we need a way to safeguard against accidental changes to the data in the raw object, as this could render the accompanying meta-data incorrect (changes to the meta data -- like raw.info -- can be considered intentional). One way to enforce this is to force raw.preload=False. 

However, @agramfort pointed out to me that even with this, it is possible to crop the raw object. It appears that MNE keeps no history of the cropping and it's impossible for us to prevent this. Would it make sense to add a private attribute (e.g., `raw._is_cropped`) to mne that can be used by mne-bids to enforce this? Any other potential snafus that we should be aware of?

agramfort, 2018-10-28:
how about rather making sure you can clone the raw object by rereading from
disk and then check that raw.times is still the same?

I would be safer I think than tracking any possible change that can have
happened to the raw object besides the info attribute.

larsoner, 2018-10-28:
Eventually it would be ideal if other operations (filtering in particular) could be done without preloading, too. And such things are feasible to implement. So whatever solution we devise should be compatible with the idea that operations on the data other than cropping could eventually be implemented, too.

agramfort, 2018-10-28:
> Eventually it would be ideal if other operations (filtering in particular)
> could be done without preloading, too. And such things are feasible to
> implement. So whatever solution we devise should be [...]

yes this also says that one should be able to reread from disk and check
that object has not changed. I don't think it's hard to do provided you
keep with attributes all the constructor parameters.

jasmainak, 2018-10-28:
okay that's fair enough. We can reload and do a check. It should be on `mne-bids` end for now I suppose?

> Eventually it would be ideal if other operations (filtering in particular) could be done without preloading, too

But this would not work with `preload=False` would it? It would be a memory mapped file, right? Otherwise you would have to modify the original file ...

larsoner, 2018-10-28:
It is possible to do it on the fly (no preload, memmap, or on disk modification), it would just be a huge pain to program

agramfort, 2018-10-29:
I would force to pass non-preloaded data and then just recreate the raw
from scratch. Think about the clone function in sklearn to clone any
estimator and get a fresh one. Makes sense?

jasmainak, 2018-10-29:
hmm ... this would not fix your cropping issue though as you would also copy `raw.times`. Or do you propose reading the `raw` back from the file? For that to happen we would need all the params to the `read_raw_xxx` files as an attribute in `raw` somewhere. It would essentially reset your `raw` to the state just after it was read in -- thus we would lose any information that the user added to `raw.info` ...

agramfort, 2018-10-30:
>
> hmm ... this would not fix your cropping issue though as you would also
> copy raw.times. Or do you propose reading the raw back from the file?
>

yes but doing this without overwritting the current raw (ant its info) and
just to check that data have still the same shape etc.

>

jasmainak, 2018-10-30:
okay sounds good. See #5679