mne-tools/mne-python #5680: Optimal transport for inverse solution validation

makkostya, 2018-10-31:
**Problem:**
We need an universal tool to compare the inverse solution to simulated source configuration, i.e. have a function f(stc_simuleted, stc_estimated) which would quantify how "close" estimated solution is to simulated one.

The existing way to compute the l2 norm between two stcs does not take into account spatial properties of sources (spatially close sources can still result in high l2 error norm.)

To solve solve this issue Dipole Localization Error (DLE) is often used, but it does not seem to me an universal solution, cause it would have issues with distributed or multiple sources configurations.   

**Possible solution:**
During several discussions of this problem, the idea of using of optimal transport was proposed.
I tried to formalize it, substantially based on H. Janati et al. "Wasserstein regularization for sparse multi-task regression", 2018. 
The main idea is to do optimal transport from estimated solution to the simulated one, and use the amount of "energy" needed for this transport as a similarity measure, i.e. smaller it is, better is the estimation. 
Please consult the latex file for mathematical formulation of this criteria, and tell me what do think about it.
@agramfort, do you have a code to compute the unbalanced Wasserstein distance that you used in this arcticle?
[OT_for_simulation_metric.pdf](https://github.com/mne-tools/mne-python/files/2534686/OT_for_simulation_metric.pdf)

agramfort, 2018-10-31:
@hichamjanati could share some code for sure and it would nice to have this. However I would first focus on more commonly used metrics. In your literature review did you list the commonly used ones? if so can you share them here?

hichamjanati, 2018-11-05:
@makkostya  You can find code for the unbalanced Wasserstein distance here https://github.com/hichamjanati/unbalanced-ot. 
Something is not clear in your document, from what I understand, you would like to include both temporal and spatial shifts in your ground metric M, but at the same time since you compute the L2 norm between time series, there is not 'transport in time' right ?  How do you compute \theta in R^Ns from your stc matrix S ?

makkostya, 2018-11-05:
@hichamjanati Yes, in the way I presented it there is no transport in time. The similarity of time series is taken into account only to define the cost matrix between sources. We could add transport in time by it will increase the dimension of the problem.

\theta_i = || S_i. ||
t_i = S_i. / || S_i. ||

Yes indeed, p = Ns