mne-tools/mne-python #5686: Return indices of rejected epochs

dpkoller, 2018-11-01:
Hi I have a request for a useful feature. I hope I haven't overlooked this but after looking through the MNE documentation I couldn't find anything that would satisfy what I and probably many people would need.

#### Describe the problem
When rejecting epochs visually using _mne.Epochs.plot()_ the marked epochs get rejected when closing the visualization. In many cases it would be desirable to reject those epochs later or have a list of indices that were rejected.

#### Describe your solution
When marking epochs visually for rejection using _mne.Epochs.plot()_ the bad epochs' indices should be returned, such that one can reject epochs after visually identifying them along with specifying a meaningful reason for rejection e.g. using the _mne.Epochs.drop()_ method. Ideally the user would be requested to enter a reason for rejection at the time of marking an epoch to be rejected.

#### Describe possible alternatives
A possible alternative would be to retain the annotations attribute and rejection_by_annotations method for mne.Epochs as well.

larsoner, 2018-11-01:
Look at `epochs.selection` and `epochs.drop_log`. Can you make a PR to the [epochs tutorial](https://github.com/mne-tools/mne-python/blob/master/tutorials/plot_object_epochs.py) (which turns into [this page](http://mne-tools.github.io/dev/auto_tutorials/plot_object_epochs.html)) to make this clearer?

dpkoller, 2018-11-01:
Hi, thanks a lot for the fast reply! So the easiest way would be to retrieve the already rejected epochs from `epochs.drop_log` by searching for the 'USER' strings and returning their indices?
Then one would have to note down the reasons for rejecting each epoch and modifying the 'USER' string, right?

larsoner, 2018-11-01:
Any non-empty list entry means it has been rejected for some reason or another. Also you should find that `np.where([len(log) == 0 for log in epochs.drop_log])[0]` is the same integer array as `epochs.selection`.

agramfort, 2018-11-01:
the epochs.selection gives you the indices of the remaining epochs.

closed 2018-11-01 by larsoner