mne-tools/mne-python #5688: stc.as_volume function not available

vivekosdd, 2018-11-02:
HI,
I'm following this tutorial: https://www.martinos.org/mne/stable/auto_examples/inverse/plot_compute_mne_inverse_volume.html
I can run it successfully with sample data, but when I run with my own data it runs successfully till stc.crop but does not show the stc.as_volume function.

agramfort, 2018-11-02:
can you tell us more about your data? what is the error?

also for problems that are not bugs in the code please use the mailing list
or the gitter channel.

vivekosdd, 2018-11-02:
My data is a resting state MEG data.
I have created the evoked and inverse_operator file.
after computing stc I do not get option stc.as_volume which is available when doing with sample data.
Error: `>>> img = stc.as_volume(src, mri_resolution=False)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'SourceEstimate' object has no attribute 'as_volume'`

agramfort, 2018-11-02:
to use as_volume you need a volume source space and not a surface based one.

see https://mne-tools.github.io/dev/auto_tutorials/plot_object_source_estimate.html#sphx-glr-auto-tutorials-plot-object-source-estimate-py
or
https://mne-tools.github.io/dev/auto_examples/inverse/plot_compute_mne_inverse_volume.html?highlight=volume

vivekosdd, 2018-11-02:
Thanks. 
To generate volume source estimate we need vertices as an input. To generate vertices I come across this function called, center_of_mass(subject=None, hemi=None, restrict_vertices=False, subjects_dir=None, surf='sphere')
when I run this command with my data input it shows the following error: `TypeError: center_of_mass() missing 1 required positional argument: 'self'`
whereas in documentation no such argument is defined.

agramfort, 2018-11-03:
you need a volume source space:

https://www.martinos.org/mne/stable/generated/mne.setup_volume_source_space.html

vivekosdd, 2018-11-05:
Using this function gives the following error.
`src_vol = mne.setup_volume_source_space(subject='vivek1', pos=5.0, mri='/home/vivek/freesurfer/subjects/vivek1/mri/T1.mgz', sphere=(0.0, 0.0, 0.0, 90.0), bem='/home/vivek/freesurfer/subjects/vivek1/bem/inner_skull.surf', surface=None, mindist=5.0, exclude=0.0, subjects_dir='/home/vivek/freesurfer/subjects/', volume_label='/home/vivek/freesurfer/subjects/vivek1/mri/aseg.mgz', add_interpolator=True, verbose=None)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<string>", line 2, in setup_volume_source_space
  File "/home/vivek/anaconda3/lib/python3.6/site-packages/mne/utils.py", line 729, in verbose
    return function(*args, **kwargs)
  File "/home/vivek/anaconda3/lib/python3.6/site-packages/mne/source_space.py", line 1545, in setup_volume_source_space
    volume_labels = get_volume_labels_from_aseg(mri)
  File "/home/vivek/anaconda3/lib/python3.6/site-packages/mne/source_space.py", line 2439, in get_volume_labels_from_aseg
    for ii in np.unique(mgz_data)]
  File "/home/vivek/anaconda3/lib/python3.6/site-packages/mne/source_space.py", line 2439, in <listcomp>
    for ii in np.unique(mgz_data)]
IndexError: index 0 is out of bounds for axis 0 with size 0`

agramfort, 2018-11-05:
I don't know. Does it work with other subjects?

vivekosdd, 2018-11-05:
I tried with other subject also, but same error.

agramfort, 2018-11-08:
does it work for you with sample dataset?

vivekosdd, 2018-11-15:
yes

agramfort, 2018-11-16:
then there must be a tiny difference between what you do and we did with
sample

larsoner, 2019-09-19:
@vivekosdd the volume label code was recently updated so it's possible this bug is fixed, feel free to try again. If it still fails, feel free to reopen (preferably with some data and a minimal script to reproduce the error)

closed 2019-09-19 by larsoner