mne-tools/mne-python #5703: [Feature request]: add to_data_frame() function to ICA, like mne.io.Raw.to_data_frame()

IDCtolik, 2018-11-10:
Hi,
I recently started working in the mne environment and looking for a tool to tabulate the data used to graph the ICA components.
The way it works with mne.io.Raw.to_data_frame() seems very convenient, I was wondering if it is possible to release a similar function for the ICA object?

dengemann, 2018-11-10:
Hi! What exactly do you want to put in a data frame? The unmixing matrix or
the ICA sources (after applying ICA to data). In the latter case take a
look at ica.get_sources() which will return the sources in an mne object
that has the to_data_frame method.

IDCtolik, 2018-11-13:
Hey,
Thank you very much for the help! worked like a charm.

closed 2018-11-13 by IDCtolik