mne-tools/mne-python #5706: [RFC] Draft for the new release note.

massich, 2018-11-12:
I put together a release note, please feel free to comment on it (or edit).

--- 

Dear community,

We are very pleased to announce the new 0.17 release of MNE-Python (http://martinos.org/mne/stable/).

A few highlights
================

- This release will be the last release compatible with Python 2. The next version will be Python 3 only.

- Better support for Annotations, including readers for EEGLAB, BrainVision, EDF, CSV, TXT and Brainstorm formats, and a new tutorial in the documentation to dive in.

- Better support to import data from FieldTrip and Neurmag 122 systems.

- Add capability to read and save Epochs containing complex data (e.g. after Hilbert-transform).

- Add optically pumped magnetometer dataset and examples.

- Add possibility to save source estimates in HDF5.

- New source morph object to unify morphing any type of source estimates (surface or volume) from one subject to another for group studies. It is now possible to do group studies when working on the volume with MNE.

- Add ability to pass a precomputed forward solution to mne.simulation.simulate_raw.

- Add ability to read and write beamformers.

- New source power spectral estimation example for restint-state data

- Better support for Reports, now they can be load/saved in HDF5 and the existing figures from a report can be removed.

- Add support for reading MATLAB ``v7.3+`` for EEGLAB.

- Add interactive visualization of volume source estimates using plot_volume_source_estimates

- New BIDS-compatible raw file name construction

- Better helmet support for Artemis123 and CTF, plus an automated MEG helmet shape approximation 

Notable API changes
===================

- Deprecation of annot and annotmap parameters in mne.io.read_raw_edf 

- mne.Epochs.save now has the parameter fmt to specify the desired format (precision) saving epoched data

- Deprecated mne.SourceEstimate.morph_precomputed, mne.SourceEstimate.morph, mne.compute_morph_matrix, mne.morph_data_precomputed and mne.morph_data in favor of mne.compute_source_morph

- Calling mne.Epochs.decimate no longer copies the data when decim=1. 

- Warning messages are now only emitted when warn_explicit is set rather than also being emitted as logging messages (unless a logging file is being used) to avoid duplicate warning messages.

- src.kind now equals to 'mixed' (and not 'combined') for a mixed source space (made of surfaces and volume grids) 

- The default value of stop_receive_thread in mne.realtime.RtEpochs.stop has been changed to True 

- Using the mne.io.Raw.add_channels on an instance with memmapped data will now resize the memmap file to append the new channels on Windows and Linux

- Mismatches in CTF compensation grade are now checked in inverse computation 

For a full list of improvements and API changes, see:

http://martinos.org/mne/stable/whats_new.html#version-0-17

To install the latest release the following command should do the job:

pip install --upgrade --user mne

As usual we welcome your bug reports, feature requests, critiques, and

contributions.

Some links:

- https://github.com/mne-tools/mne-python (code + readme on how to install)

- http://martinos.org/mne/stable/ (full MNE documentation)

Follow us on Twitter: https://twitter.com/mne_news

Regards,

The MNE-Python developers

People who contributed to this release (in alphabetical order):

* Alex Gramfort
* Christian Brodbeck
* Daniel McCloy
* David Sabbagh
* Denis A. Engemann
* Dirk Gütlin
* Eric Larson
* Eric Larson
* Erkka Heinila
* Ezequiel Mikulan
* Henrich Kolkhorst
* Jasper van den Bosch
* Joan Massich
* Jona Sassenhagen
* Luke Bloy
* Luke Bloy
* Mainak Jas
* Marcin Koculak
* Marijn van Vliet
* Peter Molfese
* Rasmus Zetter
* Sheraz Khan
* Stefan Appelhoff
* Stefan Repplinger
* Steven Bethard
* Steven Gutstein
* Teon Brooks
* Thomas Hartmann
* Tommy Clausner

larsoner, 2018-11-12:
Edited, LGTM

cbrnr, 2018-11-12:
I guess I contributed, unless there is a minimum amount of code to be included in the list?

larsoner, 2018-11-12:
> unless there is a minimum amount of code to be included in the list?

1 line :)

@massich how did you make the list? Following the wiki I can get:
```
larsoner@bunk:~/python/mne-python$ git rev-list -n 1 v0.16 | xargs -I commit git log commit.. | git shortlog -s -n | sed "s/^[ \t]*/* /"
* 129	Eric Larson
* 35	Joan Massich
* 28	Alexandre Gramfort
* 24	Jona Sassenhagen
* 23	Stefan Appelhoff
* 12	Clemens Brunner
* 8	Mainak Jas
* 8	Marijn van Vliet
* 5	Thomas Hartmann
* 3	Denis A. Engemann
* 3	Henrich Kolkhorst
* 3	Luke Bloy
* 3	Teon Brooks
* 2	Antoine Gauthier
* 2	Britta Westner
* 2	Christian Brodbeck
* 2	Thomas Jochmann
* 1	Daniel McCloy
* 1	DavidSabbagh
* 1	Jasper J.F. van den Bosch
* 1	Jen Evans
* 1	Johan van der Meer
* 1	Kambiz Tavabi
* 1	Lorenz Esch
* 1	MJAS1
* 1	Marcin Koculak
* 1	Peter J. Molfese
* 1	Sam Perry
* 1	Sergey Antopolskiy
* 1	Sheraz Khan
* 1	Stefan Repplinger
* 1	Steven Bethard
* 1	Teekuningas
* 1	Tom Dupré la Tour
* 1	Tristan Stenner
* 1	buildqa
* 1	ezemikulan
* 1	hubertjb
* 1	jeythekey
* 1	sarasommariva
```
And @cbrnr is listed. We should probably adapt this to produce an alphabetized list of names since this is what we show now instead of commit counts.

larsoner, 2018-11-12:
Here, I'll clean up this list and update the wiki:
```
larsoner@bunk:~/python/mne-python$ git rev-list -n 1 v0.16 | xargs -I commit git log commit.. | git shortlog -s | sed "s/^\s*[0-9]*[ \t]*/- /"
- Alexandre Gramfort
- Antoine Gauthier
- Britta Westner
- Christian Brodbeck
- Clemens Brunner
- Daniel McCloy
- DavidSabbagh
- Denis A. Engemann
- Eric Larson
- Henrich Kolkhorst
- Jasper J.F. van den Bosch
- Jen Evans
- Joan Massich
- Johan van der Meer
- Jona Sassenhagen
- Kambiz Tavabi
- Lorenz Esch
- Luke Bloy
- MJAS1
- Mainak Jas
- Marcin Koculak
- Marijn van Vliet
- Peter J. Molfese
- Sam Perry
- Sergey Antopolskiy
- Sheraz Khan
- Stefan Appelhoff
- Stefan Repplinger
- Steven Bethard
- Teekuningas
- Teon Brooks
- Thomas Hartmann
- Thomas Jochmann
- Tom Dupré la Tour
- Tristan Stenner
- buildqa
- ezemikulan
- hubertjb
- jeythekey
- sarasommariva
```

agramfort, 2018-11-13:
did you already update the mailmap?

massich, 2018-11-13:
@larsoner 	I did not know about the wiki. I copy pasted all the changes in this release in whats new, some regex magic, got a list of all the authors and then I unique them. 

But for sure your list is fairer. I'll take it into account next time.

larsoner, 2018-11-13:
I partially updated the mailmap in my decoding doc PR. Waiting to hear back from the last couple of people then it should hopefully be a list of actual names

sappelhoff, 2018-11-20:
+1 for the release notes.

What does the `.mailmap` do? (Wondering if I should add myself there)

closed 2018-11-20 by larsoner

sappelhoff, 2018-11-20:
hahaha great! And while we are at it, why is the `AUTHORS.rst` so limited?

larsoner, 2018-11-20:
We should probably just remove it. It does not get updated and I don't think it's used or useful

agramfort, 2018-11-20:
yes we are sloppy with AUTHORS.txt ...