mne-tools/mne-python #5708: Compute TFR with fmin=0Hz

cbrnr, 2018-11-13:
Currently, `tfr_morlet` and `tfr_multitaper` throw an error if the minimum frequency is 0Hz:

```
/usr/local/lib/python3.7/site-packages/mne/time_frequency/tfr.py in morlet(sfreq, freqs, n_cycles, sigma, zero_mean)
     83         # this scaling factor is proportional to (Tallon-Baudry 98):
     84         # (sigma_t*sqrt(pi))^(-1/2);
---> 85         t = np.arange(0., 5. * sigma_t, 1.0 / sfreq)
     86         t = np.r_[-t[::-1], t[1:]]
     87         oscillation = np.exp(2.0 * 1j * np.pi * f * t)

ValueError: arange: cannot compute length
```

and

```
/usr/local/lib/python3.7/site-packages/mne/time_frequency/tfr.py in _make_dpss(sfreq, freqs, n_cycles, time_bandwidth, zero_mean)
    140 
    141             t_win = this_n_cycles / float(f)
--> 142             t = np.arange(0., t_win, 1.0 / sfreq)
    143             # Making sure wavelets are centered before tapering
    144             oscillation = np.exp(2.0 * 1j * np.pi * f * (t - t_win / 2.))

ValueError: arange: cannot compute length
```

FWIW, `tfr_stockwell` works fine with 0Hz.

Maybe I'm totally misunderstanding something, but I think it would be nice if TFRs started at DC, independent of the TFR method.

MWE to reproduce:

```
import numpy as np
import mne

sfreq = 128
n_epochs, n_ch, n_time = 144, 2, 1921
data = np.random.randn(n_epochs, n_ch, n_time)
info = mne.create_info(n_ch, sfreq, 'eeg')
epochs = mne.EpochsArray(data, info)

freqs = np.arange(0, 4)
mne.time_frequency.tfr_morlet(epochs, freqs, n_cycles=freqs / 2)  # throws error
mne.time_frequency.tfr_multitaper(epochs, freqs, n_cycles=freqs / 2)  # throws error
mne.time_frequency.tfr_stockwell(epochs, fmin=0, fmax=3)  # works
```

agramfort, 2018-11-13:
+1 to get a meaningful error but it does not make sense to look  at such
low freqs with short signals.

>

cbrnr, 2018-11-14:
While it may not make much sense to interpret the DC component of an EEG spectrum, technically this should be well defined - it is the mean of the signal. Also, `tfr_stockwell` can deal with 0Hz components and I don't see why the other two FFT-based methods don't work.

agramfort, 2018-11-14:
sure PR welcome

DiGyt, 2019-04-17:
@cbrnr @agramfort 

I could work on this if you want.
So basically, both of these errors are in fact ZeroDivisionErrors, created if ``` f = 0``` in ```sigma_t = this_n_cycles / (2.0 * np.pi * f)``` and ```this_n_cycles / float(f)```.

So the options probably would be:
1. Print a meaningful error like ```raise ValueError("All frequencies must be greater than 0")```
2. Mask the case f = 0 with something like f = 1e-300 (Which worked for me practically, anyhow, i'm not sure how you feel about this)

Setting a minimum frequency of 0 to DC offset (if i understood your intention right?) would not solve the particular case if DC offset is also 0.

DiGyt, 2019-04-17:
BTW, here's how this Error occurs:
```python
import numpy as np
sfreq = 128
this_n_cycles = 0.0

f = 0
sigma_t = this_n_cycles / (2.0 * np.pi * f)	# ZeroDivisionError: float division by zero
t = np.arange(0., 5. * sigma_t, 1.0 / sfreq)

f = 1e-300
sigma_t = this_n_cycles / (2.0 * np.pi * f)
t = np.arange(0., 5. * sigma_t, 1.0 / sfreq)	# passes

# this is the case in mne/time_frequency/tfr.py::morlet()
f = np.int64(0)
sigma_t = this_n_cycles / (2.0 * np.pi * f)	# sigma_t = nan
t = np.arange(0., 5. * sigma_t, 1.0 / sfreq)	# ValueError: arange: cannot compute length
```

cbrnr, 2019-04-17:
I'd prefer an error message if it is technically not possible to compute the DC component.

DiGyt, 2019-04-18:
> I'd prefer an error message if it is technically not possible to compute the DC component.

Well, it surely is technically possible to do so, but seems not to be directly related to the issue that morlet and multitaper can't be computed with fmin=0Hz.

Was your intention to remove DC offset from each channel by subtracting the mean before calculating the TFR, or would you like a frequency ```freqs = [0]``` to be corrected to DC each time it is passed?

cbrnr, 2019-04-18:
I would like a frequency of 0 to mean DC so that all TFRs behave like `tfr_stockwell` (or any other FFT-based method, where 0Hz does not produce an error).

DiGyt, 2019-04-24:
ok, so i listed some examples below, where i manually set ```freq = 0``` to DC (outside the function). As you can see, this would not solve the problem for all cases.

```python

import numpy as np
import mne

sfreq = 128
n_epochs, n_ch, n_time = 1, 1, 2048
info = mne.create_info(n_ch, sfreq, 'eeg')
freqs = np.arange(0, 4)

# 1. case with small dc

data1 = np.random.randn(n_epochs, n_ch, n_time)		# random channel
epochs1 = mne.EpochsArray(data1, info)
dc1 = np.mean(epochs1._data)
freqs1 = np.where(freqs==0, dc1, freqs)			# replace freq == 0 with dc
tfr1, _ = mne.time_frequency.tfr_morlet(epochs1,
					freqs1, n_cycles=freqs1 / 2)	# this works
				
tfr1.freqs					# of course now freqs would be inconsistent

# 2. case with large dc

data2 = np.random.randn(n_epochs, n_ch, n_time) + 2	# introduce high offset
epochs2 = mne.EpochsArray(data2, info)
dc2 = np.mean(epochs2._data)
freqs2 = np.where(freqs==0, dc2, freqs)			# replace freq == 0 with dc
tfr2, _ = mne.time_frequency.tfr_morlet(epochs2,
					freqs2, n_cycles=freqs2 / 2)	# this works

tfr2.freqs					# but this will be extremely confusing

# 3. case in which dc == 0

data3 = np.ones([n_epochs,n_ch,n_time])
data3[:,:,::2] = -1					# oscillating channel with mean = 0
epochs3 = mne.EpochsArray(data3, info)
dc3 = np.mean(epochs3._data)
freqs3 = np.where(freqs==0, dc3, freqs)			# replace freq == 0 with dc
tfr3, _ = mne.time_frequency.tfr_morlet(epochs3,
					freqs3, n_cycles=freqs3 / 2)	# this still won't work

```
That's the reason i'm kind of hesitating with this, since doing it would not solve ALL the cases, while simultaneously introducing inconsistencies in the frequency intervals.

The reason that 0 Hz frequencies can be passed into stockwell but not multitaper or morlet, is (in my understanding) only based on the fact that in multitaper & morlet, we come along a ```xy/ freq``` situation, that results in a ZeroDivisionError, while in stockwell and ffts we don't. Therefore a direct solution to this would rather be to try and deal with these fractions than setting fmin to dc.