mne-tools/mne-python #5709: Deprecate stim channel

massich, 2018-11-13:
i.e. EDF rader synthetize the stim channel so it should warn.
```py
import os.path as op
import mne
edf_path = op.join(op.dirname(op.dirname(mne.__file__)), 'mne/io/edf/tests/data/test.edf')
raw_py = read_raw_edf(edf_path, misc=range(-4, 0), stim_channel=139, preload=True)
```

massich, 2018-11-15:
closing via #5712

massich, 2018-11-16:
I just came across this:
https://github.com/mne-tools/mne-python/blob/57272be1d342e1058670ebac0e3ab292deb56767/mne/io/egi/egimff.py#L220

egi files also have a synthetic stim channel that was not using the utils for synthetizing stim channels. 

Can anyone comment on this?

cc(based on `./mne/io/egi` blame ):@agramfort, @larsoner, @adykstra , @drammock , @dengemann , @fraimondo , @jasmainak , @wmvanvliet , @teonbrooks

larsoner, 2018-11-16:
I'll open a PR to deprecate that, too

agramfort, 2018-11-16:
oh yes we had forgotten about this one :( :( we need to apply the same
logic to this one too....

larsoner, 2018-11-16:
Actually maybe we should deprecate... the EGIMFF has stim channels but users can use those directly (i.e., don't also synthesize to STI 014). The EGI does seem to have "event codes" like in BV.

We first need to change these event codes over to Annotations, that step should be done first before deprecating the `stim_channel` creation. @massich can you do that since you've been doing that for the other readers?

agramfort, 2018-11-16:
would be great to finish this before the release

fraimondo, 2018-11-16:
The EGI system allows for digital inputs as triggers, but it creates on channel per value. So you might endu p with 255 digital channels (D001-D255). Also, you can add manual “marks”.

The currently generated STI 014 does not reflect the original values of the digital inputs, so I still do not know what is the meaning of this channel.

Example:
```
In [7]: events = mne.find_events(raw)
2048 events found
Event IDs: [ 1  2  3  4  5  6  7  8  9 10 11 12 13]

In [8]: [x for x in raw.ch_names if x.startswith('D')]
Out[8]: ['D190',  'D255',  'D126',  'D102',  'D158',  'DI94', 'DI70', 'D180', 'D116', 'D108', 'D148', 'DI84', 'DI76']
```

agramfort, 2018-11-16:
@fraimondo you are our expert here so please tell us how to do it right

cc @deep-introspection

fraimondo, 2018-11-16:
If you still want to synthesise the STI 014 channel, then it's a matter of looking at the DXXX channels, multiplying it's content by 255-XXX and add them all up.

Something like that:
```Python
dnames = [x for x in raw.ch_names if x.startswith('D')]
values = [int(x.replace('DIN', '').replace('DI', '').replace('D', '')) for x in dnames]
values = 255 - np.array(values)
stim_data = np.zeros_like(raw._data[raw.ch_names.index(dnames[0]), :])

for dchan, dvalue in zip(dnames, values):
    if dvalue == 0:
        continue
    ddata = raw._data[raw.ch_names.index(dchan), :]
    idx = np.where(ddata != 0)[0]
    stim_data[idx] = dvalue
````

agramfort, 2018-11-17:
if EGI / MFF works with stim channels and not onset/description that maps
to Annotations structure maybe it's not worth the pain...

thoughts from our EGI/MFF users?

dengemann, 2018-11-17:
Cc @deep-introspection

larsoner, 2018-11-17:
Also it occurred to me that if we have to deprecate this EGI stim in 0.18
instead of 0.17 it's probably okay, too

>

agramfort, 2018-11-17:
yes. If we have no news or suggestion on this by monday and let's do it
later and move on with the release

deep-introspection, 2018-11-17:
Indeed, the order of the stim influence the event ids from what I remember. The ideal would be to have a mapping of those events id with the name of triggers which are supposed to be stored in the .MMF file (EGI software rename them properly).

agramfort, 2018-11-17:
So is there anything we should fix or improve. ? Can you share code that shows some unnecessary pain ?

dengemann, 2018-11-17:
As far as I understand understanding one’s mmf file is key. The labels for
the event channels unfortunately depend on the operator and can easily
change from session to session. By enumerating the events it often happens
that different events land on a channel with the same name (across
recordings). Once you are in the raw format there is little to do.
Essentially you have to know what is called how.

fraimondo, 2018-11-18:
> @fraimondo <https://github.com/fraimondo> - the last time I used mne synthesized channel for .raw files it coded the event values according to the order of appearance. So not along the DIN values - and I always used a reading scheme similar to the code you pasted.

And there is also the manual “marks”, which come with a text value and are customisable by the user. So I guess the MFF format has both a stim channel and annotations. And as it is done, it can be the case that the same sample has two events.

Or we either fix the stim channel or we remove it. I never created a PR because I assumed someone was using the STI 014 channel as generated, and this will create  a conflict. Indeed I just checked some old code I have to read EGI RAW files, and we use the STI 014 channel and the event ids.

It is quite hard to find a solution that fits every case.

agramfort, 2018-11-18:
question is can we avoid synthesizing a stim channel when not present in
the raw file just by using annotations

unless there is a strong energy now to look into this in the next days
we're going to release and postpone this.

dengemann, 2018-11-18:
I’d postpone it. This one is not easy to fix. We could perhaps see where
the doc can be improved right now and call for more input from egi users on
the mailinglist.

cbrnr, 2019-08-30:
What's the status on this issue? I remember having discussed this somewhere else, but I can't seem to find the thread. I was arguing for keeping the `stim_channel` argument because people still use analog channels to record their triggers (but by default, no stim channel should be synthesized, this should only happen upon explicit user request).

agramfort, 2019-08-30:
Yes I think it's what we did and agreed on.

>

cbrnr, 2019-08-30:
OK, so close this issue then?

closed 2019-08-30 by agramfort