mne-tools/mne-python #5719: raw.get_data() takes same amount of time on 1 channel as on 64 channels

russellu, 2018-11-16:
Hello, i am trying to load EEG data using the following:

`montage_path = '/media/sf_shared/standard-10-5-cap385.elp'
montage = mne.channels.read_montage(montage_path)
raw = mne.io.read_raw_brainvision(
        raw_eeg_path,montage=montage,eog=['ECG','ECG1'])

dec = np.load(data_path)
raw.get_data(picks=[0])`

however, if i leave picks blank (so it loads all channels), it takes the same amount of time to load as if i only pick one (there are 64 channels). The memory requirements, however, are less for just one channel (As would be expected) is this a bug, or is it not possible to load brainvision files by single channel?

jona-sassenhagen, 2018-12-17:
@russellu anything else on this issue, or can it be closed?

mmagnuski, 2018-12-19:
I'd leave it open for a while - someone (@russellu ?) could post their timings and someone else might investigate a bit.

jona-sassenhagen, 2019-03-24:
@russellu ?

sappelhoff, 2022-05-26:
I cannot reproduce this:

```python
# %%
import mne

testing = mne.datasets.testing.data_path()
vhdr = testing / "montage" / "bv_dig_test.vhdr"
raw = mne.io.read_raw_brainvision(vhdr, preload=False)

# %%
# on my machine: 511 µs ± 50.8 µs per loop
%%timeit
raw.get_data(picks=[0])

# %%
# on my machine: 3.87 ms ± 222 µs per loop
%%timeit
raw.get_data()
```

the testing data shipped with MNE-Python is rather small in size (around 1MB), but I also found the same when using my own data (around 280MB).