mne-tools/mne-python #5721: mne does not return event labels using find_events or _read_vmrk_events

russellu, 2018-11-18:
Hello, i am trying to load all the events for a given brainvision dataset (.vhdr, .eeg, .vmrk). however, i haven't been able to get the actual event labels using any of the functions i've found so far, and it drops any event that isn't prefixed by "Response"

my marker file (.vmrk) has a total of 5127 events, each with string labels/latencies, here are examples of 3 types of events in my .vmrk:
1)
Mk3382=Response,R1,36568476,1,0
2)
Mk3395=Bad Interval,BadMin-Max,13132,187,0
3)
Mk5011=Scoring,NREM2,960000,0,0

mne.find_events: returns a 3391 x 3 array, so it returns all the R1 markers (example #1) but ignores the Bad Interval and Scoring markers.

brainvision._read_vmrk_events: returns a 3630 x 3 array, it returns all the R1 and Scoring markers, but ignores the Bad Interval markers.

is there a way to simply load all the events, to get an array with the trigger label and latency? from there, i can find the markers i need.

russellu, 2018-11-18:
i should note that some of these events were added in after the scan, as scoring evens (ie, the Bad Interval and Scoring events). not sure if that matters.

agramfort, 2018-11-18:
what version of MNE are you using?

do you see them in raw.annotations provided you use current master?

agramfort, 2018-11-18:
ok you need to try the current nightly version. We have significantly
improved support of marker/annotations for BrainVision files.

can you try and tell us how it goes?

thanks

hoechenberger, 2021-02-14:
This was probably fixed long ago.

closed 2021-02-14 by hoechenberger