mne-tools/mne-python #5722: RuntimeError: Reader error, notify mne-python developers

usarawgi911, 2018-11-18:
Firstly, thank you for this amazing library, helps a lot. :))

[image]

I am running into the above error using MNE 0.16.2 -- "RuntimeError: Reader error, notify mne-python developers" originating from mne/io/base.py (as can be seen in the above snapshot as well)

Any help with be highly appreciated :))

agramfort, 2018-11-19:
you use openbci data.

You should report the issue here: https://github.com/OpenBCI/OpenBCI_MNE

usarawgi911, 2018-11-19:
Yup, I already raised the issue there but nvm, I just wrote some custom code to parse through the file and get the data into an MNE object, works. Thanks anyway :))

closed 2018-11-19 by usarawgi911

agramfort, 2018-11-19:
if you have the fix you should open a PR to fix it upstream