mne-tools/mne-python #5728: mne coreg crashes when changing MRI Subject

thht, 2018-11-20:
#### Describe the bug
mne coreg crashes when a new MRI Subject is selected

#### Steps and/or code to reproduce
1. open the coregistration GUI via the command line or via python.
2. the GUi comes up and shows the first subject in the Freesurfer subject folder or the subject that was chosen via the -s command line option.
3. select any other MRI from the dropdown list or choose a fif file that makes the GUI load either the average or an individual MRI.

#### Expected results
The selected MRI is loaded and is displayed.

#### Actual results
The GUI crashes. the whole traceback is rather long: https://pastebin.com/WRmjF3DE

i tried this on two different computers, one with an NVIDIA, the other one with an intel graphics adapter.

#### Additional information
Platform:      Linux-4.9.0-8-amd64-x86_64-with-debian-9.6
Python:        3.6.6 |Anaconda, Inc.| (default, Oct  9 2018, 12:34:16)  [GCC 7.3.0]
Executable:    /home/th/.pyenv/versions/miniconda3-latest/envs/mne_test/bin/python
CPU:           : 12 cores
Memory:        15.6 GB

mne:           0.17.0
numpy:         1.15.4 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.1.0
matplotlib:    3.0.1 {backend=Qt5Agg}

sklearn:       0.20.0
nibabel:       2.3.1
mayavi:        4.7.0.dev0 {qt_api=pyqt5, PyQt5=5.9.2}
cupy:          Not found
pandas:        0.23.4
dipy:          0.14.0

thht, 2018-11-20:
yeah, i had successfully suppressed those memories from times long-gone when i had to code in C++....

anyhow, i think that you are right assuming that it is some VTK/Mayavi issue. it is definitely not something related to the 0.17 release because i started to notice this issue yesterday when my production env was still on 0.16.2. what i had recently done though was updating the environment according to the environment.yml of this project. so vtk/mayavi might have gotten updated in the process.

so, maybe we might get lucky identifying the problematic version and sticking them to working ones?

in any case: is there any possibility to make a unit test out of this issue?

larsoner, 2018-11-20:
> in any case: is there any possibility to make a unit test out of this issue?

You can look in `mne/gui/tests/`. In principle you should be able to mimic the crash with commands that do the same things that your clicking did. But in practice you are chasing down a memory problem that might occur only in specific contexts (and we might not mimic those perfectly in tests / when doing `pytest ...`). But it's worth a shot!

Hellg0re, 2019-04-26:
In our lab, we can reproduce this behavior as well. On Windows and on Linux machines.

This Machine below has a Radeon graphics adapter installed: (The Linux machine runs with NVIDIA)
Platform:      Windows-7-6.1.7601-SP1
Python:        3.7.2 (default, Jan  2 2019, 17:07:39) [MSC v.1915 64 bit (AMD64)]
Executable:    C:\Users\...\Anaconda3\envs\mne\pythonw.exe
CPU:           Intel64 Family 6 Model 85 Stepping 4, GenuineIntel: 12 cores
Memory:        31.7 GB

mne:           0.17.0
numpy:         1.15.4 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.2.0
matplotlib:    3.0.2 {backend=Qt5Agg}

sklearn:       0.20.2
nibabel:       2.3.3
mayavi:        4.7.0.dev0 {qt_api=pyqt5, PyQt5=5.9.2}
cupy:          Not found
pandas:        0.24.0
dipy:          0.15.0

larsoner, 2019-05-16:
This was fixed in #6303

closed 2019-05-16 by larsoner