mne-tools/mne-python #5732: Add more actionable headers to the Annotations tutorial page

choldgraf, 2018-11-20:
Hey all - I recently had to learn how annotations work in MNE, in relation to events arrays. A couple quick thoughts on the [annotations tutorial](https://www.martinos.org/mne/stable/auto_tutorials/plot_object_annotations.html).

1. We should have a section that shows how to do events -> annotations -> events round-trip.
2. I'd recommend using more actionable headers on that page. E.g. rather than "Working with annotations" make the headers a specific thing someone wants to do, e.g. "Add annotations to a Raw object", "Find annotations from a file", etc. (or, just add sub-headings with these actionable things). IME usually people parse the docs with a specific thing in mind that they want to do, they don't usually read through the whole thing top to bottom.
3. Be a bit more clear how people should handle events/annotations in non-FIF file import. Maybe that's in another page in the docs, but I didn't easily discover it. This is particularly important since now there are a bunch of deprecation warnings around event handling in the `.io` module

Just a few thoughts I wanted to write down before I forgot!

massich, 2018-11-20:
for completness, this issue cross-reference with #5201

agramfort, 2018-11-21:
thanks a lot for the feedback

yes it would be great to further improve this first attempt of tutorial.

PRs are more than welcome :)

choldgraf, 2018-11-21:
I know I know, I just wanted to get the ideas down while they were in my brain :-)  maybe we can make a docs push at the next sprint ;-)

massich, 2019-05-15:
@drammock can you check that we addressed #6307 (and close if so)? 

thx

drammock, 2019-05-15:
Some of these suggestions are WIP-implemented in #5944, in a [new annotations tutorial[(https://12627-1301584-gh.circle-artifacts.com/0/dev/auto_tutorials/continuous_data/35_annotations.html) and a [new events tutorial](https://12627-1301584-gh.circle-artifacts.com/0/dev/auto_tutorials/continuous_data/40_events.html).  Based on comments here and in #5201 I've added a few TODO checkboxes to the #5944 roadmap under "events" ...  @choldgraf if you have time you could look at those two WIP tutorials and mention in #5944 what else you think is missing from them (other than the TODOs already added to the roadmap).

closed 2019-05-15 by drammock