mne-tools/mne-python #5735: ENH topomap validation and bad channels

tomMoral, 2018-11-21:
When using `mne` to plot topomaps computed with some external methods, it highlights a channel that appears to be "bad", in the sense that the topomap is discontinuous:

[image]

Here is a snippet reproducing this topomap with files from [numpy_files.zip](https://github.com/mne-tools/mne-python/files/2603727/numpy_files.zip):
```python
import numpy as np
import mne

pos = np.load("pos.npy")
u_hat = np.load("spatial_map.npy")

mne.viz.plot_topomap(u_hat, pos)
```

I would be interested in having a utility function returning the spatial regularity of a topomap and the potential bad channels. And I think it would be better in mne as you need to correctly parse the position of each sensor to compute the spatial differential.

agramfort, 2018-11-21:
is this a plotting bug or you really have a bad channel channel in this smooth topo? one way to quantify smoothness would be the look at spectrum (spatial). You could compute the eigenvectors of the laplacian of the graph of sensors .... an alternative would be to look at dipolarity with a forward operator.

tomMoral, 2018-11-21:
I don't think it is a plotting issue as I have not seen that with experiments on `somato`. But it might be a bug in the sensor mapping from `mne-hcp`.

I would avoid dipolarity for that as we might have maps with non-dipolar patterns, _e.g._ topomaps of auditory evoked responses.

Do you have something in mind to compute the graph of sensors? Or do you mean the complete graph? Or just some nearest neighbor graph?

I was thinking of interpolating each sensors based on all other and compare the value. Then you could see if some sensors are way off compared to the other.

agramfort, 2018-11-21:
> I don't think it is a plotting issue as I have not seen that with experiments on somato. But it might be a bug in the sensor mapping from mne-hcp.

can we check? I would be surprised though

> I would avoid dipolarity for that as we might have maps with non-dipolar patterns, e.g. topomaps of auditory evoked responses.

ok

> Do you have something in mind to compute the graph of sensors? Or do you mean the complete graph? Or just some nearest neighbor graph?

graph of neighbors

see for example :
https://journals.plos.org/plosone/article?id=10.1371/journal.pone.0121741

> I was thinking of interpolating each sensors based on all other and compare the value. Then you could see if some sensors are way off compared to the other.

that's another option.