mne-tools/mne-python #5740: [CI] Broken Cricle CI

massich, 2018-11-22:
Since the merge of #5713 Circle is broken. I'm not sure if the problem is in `plot_point_spread` or just a random loky error.

see: https://circleci.com/gh/mne-tools/mne-python/10301?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link

I've retriggered the builds, let's see what happens.

massich, 2018-11-22:
+ we have all this vtk business. Does anyone know how can we fix it?

cbrnr, 2018-12-19:
CircleCI still (again?) fails on master with this message:

    Username for 'https://github.com': fatal: could not read Username for 'https://github.com': No such file or directory

larsoner, 2018-12-19:
Should be fixed by 2926807dd65d89205b91677362bf19c8d7c6de5b but if not I'll keep pushing until it does :)

closed 2018-12-19 by larsoner