mne-tools/mne-python #5744: Calculate and visualize for two caps the compute all-to-all connectivity in sensor space

AmirDJV, 2018-11-22:
Hi all, I have an EEG data which was collected from two caps/heads at the same time (hyper scanning). The data comes as a single file with 64 channels (first 32 channels are for subject 1, second 32 channels are for subject 2). While I’m doing the pre-processing for each cap alone, my final goal is to compare them. More specifically, I’m trying to calculate and visualize for the two caps the [compute all-to-all connectivity in sensor space.](https://martinos.org/mne/dev/auto_examples/connectivity/plot_sensor_connectivity.html#sphx-glr-auto-examples-connectivity-plot-sensor-connectivity-py)

My attempts were to combine the two subjects after cleaning by creating a new raw object. When I build the `info` for the new raw, understandably, I get an error for duplicate names – which only make sense as the caps identical. Thus, the` mne.create_info()` automatically number them to 0 and 1 (e.g. Fp1-0 and Fp1-1). While it solves me this issue, it causes other issues that I don’t know how to deal with.

1 – Setting montage – because I have duplicate names the montage cannot be set. 
2 – Referencing – How can I reference each cap to his own average? I looked into `set_eeg_reference()`, have I missed it? 
3 – Sensors location and presentation – because the montage cannot be set, I’m unable to visualize it.  Moreover, how can I visualize the two caps with their connectivity next to each other. 

Note – While I can calculate the connectivity for the 64 electrodes with incorrect names, I have the problem of identifying close physical channels, and I still have the issue of presenting the two caps next to each other with connectivity. 

Thank you very much!

agramfort, 2018-11-24:
>
> I would try to get an info that correspond to 1 cap and that contains
> montage.

then you can make new containers using the info above.

sorry  if I cannot be more specific but we would need some data
to make a tutorial for such a usecase.

deep-introspection, 2018-11-24:
@dengemann We will visioconf with @AmirDJV next Monday ;) We will keep this updated.

AmirDJV, 2018-11-24:
Thank you all for your help!

Meanwhile, I uploaded a notebook with my questions, problems, and sample data to [here](https://github.com/AmirDJV/github-question/blob/master/Two_Caps_Compute_all_to_all_connectivity_in_sensor_space.ipynb). 

@deep-introspection see you soon ;)

deep-introspection, 2018-11-26:
Here is what I got:

```
from mpl_toolkits.mplot3d import Axes3D
from copy import copy

locations = copy(np.array([ch['loc'] for ch in combined.info['chs']]))
cap1_locations = locations[:31, :3]
print("Mean: ", np.nanmean(cap1_locations, axis=0))
print("Min: ", np.nanmin(cap1_locations, axis=0))
print("Max: ", np.nanmax(cap1_locations, axis=0))

translate = [0, 0.5, 0]
rotZ = np.pi

cap2_locations = copy(cap1_locations)
newX = cap2_locations[:, 0] * np.cos(rotZ) - cap2_locations[:, 1] * np.sin(rotZ)
newY = cap2_locations[:, 0] * np.sin(rotZ) + cap2_locations[:, 1] * np.cos(rotZ)
cap2_locations[:, 0] = newX
cap2_locations[:, 1] = newY
cap2_locations = cap2_locations + translate

fig = plt.figure(figsize=[20, 10])
ax = fig.add_subplot(111, projection='3d')

#ax.scatter(cap1_locations[:, 0], cap1_locations[:, 1], cap1_locations[:, 2], marker='s', color='b', s=20)
#ax.scatter(cap2_locations[:, 0], cap2_locations[:, 1], cap2_locations[:, 2], marker='o', color='r', s=20)

for label, x, y, z in zip(combined.info["ch_names"][:31], cap1_locations[:, 0], cap1_locations[:, 1], cap1_locations[:, 2]):
    if not(np.isnan(x)):
        ax.text(x, y, z, label, color='blue')
        
for label, x, y, z in zip(combined.info["ch_names"][31:], cap2_locations[:, 0], cap2_locations[:, 1], cap2_locations[:, 2]):
    if not(np.isnan(x)):
        ax.text(x, y, z, label, color='red')

ax.set_xlim(-0.2, 0.2)
ax.set_ylim(-0.2, 0.8)
ax.set_zlim(-0.1, 0.3)
ax.set_xlabel('X axis')
ax.set_ylabel('Y axis')
ax.set_zlabel('Z axis')
ax.view_init(30, 0)

for idx, ch in enumerate(combined.info['chs']):
    new_sensor_position = locations[idx%31, :]
    ch['loc'] = new_sensor_position
    ch['loc'][:3] = np.concatenate([cap1_locations, cap2_locations])[idx, :]
    print(ch['loc'])
```

Does it work for you @AmirDJV ?

AmirDJV, 2018-11-27:
@deep-introspection thank you for your help!. 

I updated the repository with my question. So, please pull for changes and for downloading the new electrode map. 

I embodied parts of your script. However, I now have two problems/thoughts: 
1.	I wasn't able to use the montage I provided on the two subjects (combined) variable but had to choose `mne.channels.read_montage("standard_1020")` in order execute the plots.
2.	Due to similar physical positions, how can a I make sure that when I remove close connection in one subject, I won't remove electrodes which are "close" but not in the same cap/subject?. For example, while the connectivity between electrodes F4-0 and FC2-0 should be ignored, electrode F4-0 and FC2-1 should not ([cap picture](https://github.com/AmirDJV/github-question/blob/master/config/workspaces/32-cap.PNG)).

This is the [end result for now](https://github.com/AmirDJV/github-question/blob/master/graphs/temp%20result.png)

deep-introspection, 2018-11-27:
@AmirDJV why do you want to "remove close connection in one subject" ? Technically, my update of the info for the combined subjects structure should prevent the F4-0 to be near FC2-1 since they are way appart spatially.

Regarding the first error, I am not sure there is any issue. Is-it not just the limits of the original 3D plot that are not adapted anymore to your new positions?

AmirDJV, 2018-11-27:
@deep-introspection I might be over-conservative. I'm assuming that close electrodes would have (relatively) high connectivity just from being close and not from "actual" connectivity. If you think it's redundant, please do tell :). Regarding your fix, I thought (and it looked like it to me) that you did it.  Just wanted to be sure. 

I think you might be right again. Is there a way to create an all-purpose fix, so it wouldn't matter which montage I would use?

deep-introspection, 2018-11-27:
@AmirDJV Indeed, volume conduction can bias functional connectivity but I never heard about such removal of neighbouring electrodes. @dengemann, does it sound legitimate to do this ?
In any case, inter-brain pairs of sensors does not have this issue ;)

Regarding the second part, I wonder if this is the (X,Y,Z) limits. Maybe the best would be to compare 3D positions given by your montage and the standard MNE 1020.

AmirDJV, 2018-11-27:
@deep-introspection I'm mainly going with theory, not practice here. As theoretically, if you get some activity in one electrode, you might get similar but lower activity in her surrounding. Sort of voltage spread ish. Maybe I'm over exaggerating (?). I'm coming from the idea that if electrodes are too close to each other, it might cause the current to flow more and more within the skin to other electrodes. 
That's ok, I wasn't worried about inter-brain :p

I will try to compare the montages and I'll upload my results :)

AmirDJV, 2018-11-28:
@deep-introspection I'm assuming that I wasn't able to plot it because: 
1. I have three electrodes ('GND', 'IO', 'REF'), which I don't have in the MNE montage.  I could exclude them. 
2. As you suspected, it seems like they scaled diffrently. For example, in the MNE montage the possition is `'TP9', array([-0.0856192, -0.0465147, -0.045707 ]))`  but in my montage for the same electrode is `'TP9', array([-74.41343183, -24.17838967, -33.21214592]))`

closed 2018-12-03 by AmirDJV