mne-tools/mne-python #5745: TFCE on 3D spatial or 4D spatio-temporal data

ilogue, 2018-11-23:
Would `spatio_temporal_cluster_1samp_test` and the other clustering functions work on 3D spatial or 4D spatio-temporal data? I'm thinking to use it for data in MNI space. Would I have to provide it with a connectivity matrix or would it take into account a "3D lattice"?

What do you think @larsoner @ktavabi ? many thanks!

mmagnuski, 2018-11-24:
I don't know about volumetric data (surface- or channel- based representations have only one spatial dimension) but in general currently cluster based tests are limited to 2d in mne.

mmagnuski, 2018-11-24:
So if you have 3 spatial dimensions and one additional you would have to "unroll" the spatial dimensions and provide connectivity for them.

agramfort, 2018-11-25:
you can work with any connectivity provided it's regular grid or you
provide a sparse matrix to define the connectivity

ilogue, 2018-11-25:
alright I'll give that a go

larsoner, 2020-02-12:
Closing in favor of #4859

closed 2020-02-12 by larsoner