mne-tools/mne-python #5746: Allow adding electrode names next to evoked plots in evoked.plot_topomap

jdpigeon, 2018-11-23:
We're using the MNE plots to great success in the [BrainWaves educational neuroscience app](https://github.com//makebrainwaves/brainwaves)

[image]

However, we have a slightly odd user experience now because the plot we generate to show ERP waveforms at all channels isn't labeled by electrode. AFAIK, there isn't any way to do this within the constraints of what MNE provides.

In a future release, it would be really nice to be able to provide a legend to this plot so that each electrode name is labeled. This, along with some other tweaks to the graphical MNE components, is something I'd love to help out with in a future MNE sprint.

agramfort, 2018-11-24:
+1

we should also return channel name with _iter_topography and pass the ch_name
to the plotting function.

any volunteer?

jona-sassenhagen, 2018-11-26:
See https://github.com/mne-tools/mne-python/pull/5568 and https://github.com/mne-tools/mne-python/issues/4746

jdpigeon, 2018-11-26:
Yes! That example in https://github.com/mne-tools/mne-python/issues/4746 would be perfect

jona-sassenhagen, 2018-11-26:
Well the PR is still very work in progress, but you can just steal the code I posted in #4746.

PR #6211 (2019-04-27): MRG +1 plot_compare_evokeds topo functionality
Another attempt. It's not really hard, but it's my 3rd attempt so ...

mne.viz.plot_compare_evokeds gains the ability to plot to a topographical layout.
This will be slow for MEG data, but is often used for ERP papers.

Not ready for reviews yet.

Closes #4746
Closes #5746
Closes #6166
Closes #6039

Redo of #5568
closed 2019-04-27 by agramfort