mne-tools/mne-python #5749: combined M/EEG and apply_inverse and extract_label_time_course

Sambami, 2018-11-27:
Hi all,

I have a brief question about how apply_inverse and extract_label_time_course handle combined M/EEG data. Are the source estimates a combination of the MEG and EEG data, or based only on the MEG data?

Thank you,
Sam

agramfort, 2018-11-27:
it's combined so inverse pb is done jointly with all available channels. It's made possible by the whitening step. See for example https://www.ncbi.nlm.nih.gov/pubmed/25541187 and https://hal-imt.archives-ouvertes.fr/hal-01183551/document

Sambami, 2018-11-28:
Thank you very much. I did not realize this; I hope it is okay that I ask a follow-up question. To provide some background: I have M/EEG (tSSS'd) data + structural MRIs of children with many epochs (~800-900). I expect the noise to be heteroscedastic. I am using MNE to extract the current values and dSPM for visualization.

- The epochs used to estimate the covariance matrix are already tSSS'd, and artifacts are rejected with ICA. I have been using the Shrunk method, but I was wondering if I should use FA instead?

I appreciate any help you can provide!

Best,
Sam

agramfort, 2018-11-28:
please the mailing list
https://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis
and not the issue tracker to ask your questions.

To know if you should use FA or Shrunk just use the builtin
cross-validation procedure.
That's the whole point of our work with @dengemann to automate this decision.

Sambami, 2018-11-28:
Apologies. Will do, thank you.

closed 2018-11-28 by Sambami