mne-tools/mne-python #5750: Converting from physical to digital values with FIF

skjerns, 2018-11-27:
For some calculations I need the original digital value that the signal was saved with for some integer calculation.

For EDF files I can calculate those from the physical and digital minimum/maximum that are saved in the header:

```python
edf_raw = mne.io.read_raw_edf(...)
dmin = edf_raw._raw_extras[0]['digital_min'][ch_idx]
dmax = edf_raw._raw_extras[0]['digital_max'][ch_idx]
[...]
m = (dmax-dmin)/(pmax-pmin) 
digital = (m * signal).astype(int)
``` 

However, FIF files do not seem to contain such information. 
Is it possible to get back to the original ADC values using FIF files?

agramfort, 2018-11-27:
you haves info['cals'] but I doubt it's enough to get the int values in the
edf files. You'll have to look at the code.

skjerns, 2018-11-28:
Nope, unfortunately not :( ``mne_raw._cals` is `[1.0]` for me.

agramfort, 2018-11-28:
Sorry I cannot help more.

maybe https://github.com/bids-standard/pyedf can help

skjerns, 2018-11-28:
Why? I don't have this problem with EDF files, as they save the digital/physical values in the header.

However, the FIF files do not have this information?
And as FIF is used as the standard format in MNE, I thought I'll ask this question here :)
But turns out, [FIF](https://www.elekta.com/diagnostic-solutions/elekta-neuromag-triux/) is also just an proprietary format, not native to MNE.

cbrnr, 2018-11-28:
AFAIK FIF is in MNE for historical reasons, not for technical ones. Since it is rather difficult to read FIF outside of MNE (in addition to what you mention in this issue), I avoid this format whenever possible. I agree that EDF export would be very nice to have - if there is no open issue please feel free to open a new one.

skjerns, 2018-11-29:
@cbrnr I'd happily write a basic writer for EDF using [pyEDFlib](https://github.com/holgern/pyedflib).
I'll open an issue about this.

I also remember seing an issue/PR discussiong integration of `pyedf`, but I can't find it anymore?

cbrnr, 2018-11-30:
Thanks. You are probably referring to #5180, where we briefly mentioned that teaming up with existing projects would be a great idea.

cf #5755