mne-tools/mne-python #5751: mixed source space and pick_ori='vector'

annapasca, 2018-11-27:
Hi all,

I tried to use `pick_ori=vector` with a mixed source space but it crashed.

Here the example script to reproduce the problem
 [plot_mixed_source_space_inverse_vector.py](https://gist.github.com/annapasca/5f292a55e2b5f152c872e6047d495c7d).

I 'd like also to come back to #4317 since it seems #4312 doesn't implement the issue of #4317

jasmainak, 2018-11-27:
@agramfort @larsoner let us know if this is interesting for you. @annapasca tells me she has code to make it work ...

agramfort, 2018-11-28:
sure this should have worked. PRs welcome

larsoner, 2023-10-01:
We test that this works nowadays so it must have been fixed at some point along the way! But feel free to reopen if I'm mistaken

closed 2023-10-01 by larsoner