mne-tools/mne-python #5753: MAINT: DRY our docstrings

larsoner, 2018-11-28:
We could vendor `doccer` from SciPy to do things like this:

https://github.com/scipy/scipy/blob/69cfe48f37f08c71bc571956b4d044e4c5107e37/scipy/ndimage/filters.py#L54-L71

It should reduce code dup without hurting readability much. I actually think it will help readability in the long run since you 1) immediately know it's a standard MNE param and 2) there is less docstring to skip through / ignore to see the actual code.

If it seems worthwhile, I can make a quick PR to do this for `verbose`, and then incorporate it in #4511 for `picks`. Eventually it could also help with `n_jobs` (two flavors probably: cuda-allowed and cpu-only) and probably other params, too. It seems like it would cut down on a lot of otherwise unnecessary code (docstring) dup in the long run, and probably wouldn't make too much rebase difficulty in the meantime.

mmagnuski, 2018-11-28:
looks cool :)

jasmainak, 2018-11-28:
Indeed, great idea !

agramfort, 2018-11-29:
having bugs in docstrings is not commonly reported and it creates some
magic and yet another thing to learn from contributors.

I am not too enthusiastic about this

larsoner, 2018-11-29:
Have you looked at #4511? It shows why it would be advantageous. (The PR would basically be done instead of a ton of copy pastes pending review.)

In the future, if we want to improve this docstring, without this proposal we have to change something in 50 or so places instead of one. Same with our verbose docstring; I would be very hesitant to improve its wording even if it needs it because replacing them all correctly would be a huge pain. And reviewing it properly similarly annoying.

In the SciPy code, it seems very simple to infer that the docstring is being filled, so the extra developer learning overhead should be very small.

But if you insist that I wear out my Ctrl and V keys I will do it. :)

jasmainak, 2018-11-29:
I think this would benefit a lot of lazy developers. For example, I recently discovered that in the documentation of [spatio_temporal_cluster_test](https://martinos.org/mne/stable/generated/mne.stats.spatio_temporal_cluster_test.html#mne.stats.spatio_temporal_cluster_test), the `tail` parameter says: "See permutation_cluster_test". No link or anything. Nobody is going to complain because there is nothing technically wrong here -- just that it increases the barrier for people to understand the function and use it.

massich, 2019-01-18:
I've no idea how this one skipped my radar. But yes +1 to this one. And it would be even better if we can link them somehow to the sphinx terms.

PR #5921 (2019-02-13): MRG: use doccer to fill docstrings
This is a proof-of-concept for #5753. Done:

- [x] updated `decorator.py` to get any bugfixes
- [x] convert `verbose` decorator itself to be a docfiller
- [x] changed two files to use the new format: `epochs.py` and `source_estimate.py`.
- [x] update all files

Posting to get comments before I replace the remaining 334 (!) instances of `verbose : ` in our codebase.

Closes #5753.
closed 2019-02-13 by larsoner