mne-tools/mne-python #5755: Export to EDF using pyEDFLib

skjerns, 2018-11-29:
**Problem**
Right now it is only possible to export data using FIF, which is not a very useful and widely used file format. I do remember seing an issue about integration of some EDF library, but at this point I can't find it again.

**Suggestion**
I'd like to make a first, basic addition to integrate [pyEDFlib](https://github.com/holgern/pyedflib) for writing EDF+ files to disk. I'm using this library since some time and have good experience with it.

However I have a few questions regarding the approach:

- Would it be okay to use an external library for saving EDF instead of re-inventing the wheel?
- Where should I insert the code? I thought to extend `BaseRaw` with a function `save_edf()`, however it would also be possible to alter `.save()` to accept different file types
- in the basic version, this would save channels (and annotations?) only as raw as they are. I have no experience with all the Evoked, Epoched, etc. and if we could save them appropriately via EDF.

**Alternatives**
Write a native EDF-writer, which at this point I do not have the time for.

agramfort, 2018-11-29:
2 reactions:

- @holgern @robertoostenveld (author of pyedf) and anyone interested in
this should team up so there is one reference implementation of EDF files
in Python

- pyEDFlib has a dependency on cython so it's not an option to use this in
mne. Packaging compiled code is too much work.

I would support adding an entry in the MNE doc to document how to write
edf+ files with mne. You could put a link to a gist you use.

skjerns, 2018-12-06:
@agramfort I've created a gist, however, I doubt where exactly I should place it.. As an example? inside the IO docs? Inside an existing doc?

agramfort, 2018-12-06:
yes I would add a link to the gist at the end of the edf section here https://github.com/mne-tools/mne-python/blob/master/doc/manual/io.rst#european-data-format-edf

cbrnr, 2018-12-12:
Closing since this was fixed in #5777, but if a pure-Python solution emerges in the future, feel free to open a new issue.

closed 2018-12-12 by cbrnr