mne-tools/mne-python #5757: data length of  tfr_morlet output

VedaHung, 2018-11-30:
Hi MNE experts,

I conducted time frequency analysis on resampled MEG data (from 1000Hz to 500 Hz).
The length of the resulting power is  2334. However, I expect that the length should be 14*500/6, which is 1667, where 14 is the total length of epoch, 500 is sampling rate and 6 is the number of cycle. It seems that the sampling rate is still 1000, rather than the new sampling rate (500). I am pretty sure that I resampled the data before time frequency analysis. Anyone has a clue why the length of the power from tfr_morlet does not respect the length of the epoch. My code and a link to download all files for the analyses are provided below.

Thanks.
Veda

# a link for all necessary files
https://drive.google.com/open?id=1RORZxE6BVWbFQ5ORVM_YgZa2SxAj9xaI

`raw = mne.io.read_raw_fif('raw.fif',  preload=True)
        raw.resample(500)
        raw.filter(0.5, None)
        #raw.filter(None, 30)

         # read events
        sr = 0.5
        events_all = numpy.loadtxt('104_spat_decode_all_correct_only.txt')
        events_all = events_all.astype(int)
        events_all[:, 0] = events_all[:, 0] * sr + 1
        ica = mne.preprocessing.read_ica("104_spat_allf-ica.fif")
        rlist = numpy.loadtxt("104_spat_ica_artifact_list", ndmin=1).astype(int)

        # Construct Epochs and compute trf
        events_sel=events_all[(events_all[:,1]==1)&(events_all[:,2]==2),:]
        event_id = {'L2': 2}
        tmin, tmax = -2, 12
        baseline = (None, 0)
        picks = mne.pick_types(raw.info, meg=True, stim=False)
        epoch2 = mne.Epochs(raw, events_sel,event_id , tmin, tmax, picks=picks, baseline=baseline,reject=dict(mag=1.5e-11), preload=True)
        ica.apply(epoch2, exclude=rlist.tolist())

        n_cycles = 6
        freqs = np.arange(4., 100., 1.) # comparable with paper
        baseline = (None, 0)
        #power, itc = tfr_morlet(epoch2, freqs=freqs, n_cycles=n_cycles, use_fft=True,return_itc=True, decim=3, n_jobs=1)
        power= tfr_morlet(epoch2, freqs=freqs, n_cycles=n_cycles, use_fft=True, decim=3,n_jobs=1,average=False,return_itc=False)

cbrnr, 2018-11-30:
I think you are probably misinterpreting `n_cycles`. This parameter does not influence the number of time points in the output (it does influence the time/frequency resolution, but not the output shape). So if your input is 7001 time samples as in your case, you will get 7001 time samples in the TFR output. This is what happens if `decim=1`. If however you set `decim=3`, you will only get a third of the original samples, namely 2334.

jona-sassenhagen, 2018-12-21:
@VedaHung does the answer solve your issue?

mmagnuski, 2019-03-03:
I think this can be closed.

closed 2019-03-03 by mmagnuski