mne-tools/mne-python #5758: computing sources in standard volume space?

russellu, 2018-11-30:
Hello, does anyone know if there is a way to compute sources in standard space (MNI, or any other template) on a volumetric grid? i would like to compute raw, continuous sources (no baseline) over a sleep study.

i have simultaneous EEG-FMRI data, and i would like to compute my sources and align the source data with the FMRI data, so i can compute simultaneous EEG-FMRI correlations in each voxel separately. 

also, due to the fact that i am localizing rather long recordings (2+hrs) i have quite a bit of data (~2 million time points at 250Hz), obviously if i localize this to a source grid of 3x3x3mm or similar it will explode my memory...i'm wondering if there is a way to do this in a more memory efficient way? or, i can just compute sources over a lower resolution grid (8x8x8mm) and then later upsample it to FMRI resolution. 

thanks,

Russell

larsoner, 2018-11-30:
>  does anyone know if there is a way to compute sources in standard space (MNI, or any other template) on a volumetric grid?

`fsaverage` is in MNI space, and you can set up a volume grid source space for it with `setup_volume_source_space`

> obviously if i localize this to a source grid of 3x3x3mm or similar it will explode my memory...i'm wondering if there is a way to do this in a more memory efficient way?

Probably but it depends on what you want to do / localize. For example source power band spectra can be done in a memory efficient manner:

http://mne-tools.github.io/dev/auto_examples/datasets/plot_opm_rest_data.html#sphx-glr-auto-examples-datasets-plot-opm-rest-data-py

larsoner, 2018-11-30:
>  so i can compute simultaneous EEG-FMRI correlations in each voxel separately.

Just realized you said what you wanted to do...

I would use `apply_inverse_epochs` with `generator=True`, and for each epoch compute the correlation. Correlatinos can be computed in a piecewise manner, this is what we use in raw data covariance matrix calculation:

https://github.com/mne-tools/mne-python/blob/master/mne/cov.py#L465-L476

If you don't want to chunk in time, technically you could "chunk in space" by computing the inverse just in individual labels at a time (`label` argument to `apply_inverse_raw`) to limit the set of sources. Then you iterate over sources. Not sure how it would work with a volume source space but it should be possible to hack it to work.

russellu, 2018-11-30:
yeah, i am basically trying to get bandpass filtered, rectified time series in the same sampling rate as FMRI (TR=2.2s)...for 1-100Hz in source space. 

i'll try fsaverage with that link you sent in your first post and let you know how that goes.

chunking in time is ok. i was probably going to do that anyways (to average out the influence of outlying time points on the correlation) and then average the correlation across all 'chunks'.

thanks

larsoner, 2019-09-19:
Closing as this seems solved, feel free to reopen if not

closed 2019-09-19 by larsoner