mne-tools/mne-python #5759: 'SourceEstimateObject' is not iterable

russellu, 2018-12-01:
Hello, i'm trying to run the source alignment and coordinate frames example: 
https://www.martinos.org/mne/stable/auto_tutorials/plot_source_alignment.html#sphx-glr-auto-tutorials-plot-source-alignment-py
however, i get the following error when i run the very first code segment of the example on the above webpage:

```
Traceback (most recent call last):

  File "<ipython-input-10-3bbd985a43ca>", line 1, in <module>
    runfile('/media/sf_shared/eegfmripy/eegfmripy/test_srcloc.py', wdir='/media/sf_shared/eegfmripy/eegfmripy')

  File "/home/brain/anaconda3/lib/python3.6/site-packages/spyder/utils/site/sitecustomize.py", line 705, in runfile
    execfile(filename, namespace)

  File "/home/brain/anaconda3/lib/python3.6/site-packages/spyder/utils/site/sitecustomize.py", line 102, in execfile
    exec(compile(f.read(), filename, 'exec'), namespace)

  File "/media/sf_shared/eegfmripy/eegfmripy/test_srcloc.py", line 63, in <module>
    verbose=True)

TypeError: 'SourceEstimate' object is not iterable
```

i also got the same error when running apply_inverse from this other example:

https://martinos.org/mne/stable/auto_tutorials/plot_mne_dspm_source_localization.html

anyone know whats wrong here? thanks

russellu, 2018-12-01:
nope, i'm on 0.16.2

i'll update to 0.17.1, thanks for the response.

russellu, 2018-12-01:
also, i'm using anaconda, because i'm on neurodebian and it doesn't work so well with spyder if i don't go through the anaconda setup route...it seems to only detect a 0.17.0 version when i do 

`conda update -c conda-forge mne 
`
https://anaconda.org/conda-forge/mne

agramfort, 2018-12-01:
0.17.0 should be enough

larsoner, 2018-12-01:
Based on the traceback it looks like you have some small error in you code, like maybe doing `stc, = ...` instead of `stc = `. If you can paste your source file somewhere we can look.

You probably shouldn't get this error on either version of MNE, but definitely should not get it on 0.17.0 -- that webpage is generated automatically by a machine running 0.17.0, so I suspect some edit crept up in your file.

closed 2023-10-01 by larsoner