mne-tools/mne-python #5761: how do I set reject argument for EEG bad epochs rejection

psychoxiatao, 2018-12-02:
I am a fresh bird for MNE. When I used EEGlab, there is a threshold for artifact rejection, such as (-80 uV, 80 uV). However, in MNE, I did not know how to set the value of the argument to reject as I did not know the units of it. I found that there is a different value in the different example code. eg. eeg=60e-6, eeg=180e-6, eeg=200e-6.

jasmainak, 2018-12-02:
hey @psychoxiatao you might want to try out autoreject. See an [example here](http://autoreject.github.io/auto_examples/plot_estimate_global_reject.html#sphx-glr-auto-examples-plot-estimate-global-reject-py) how the threshold is learned automatically from the data.

agramfort, 2018-12-02:
the units of reject is always the unit of the channel type so for EEG it's
Volt. To match what you did in EEGLAB it would be reject=dict(eeg=160e-6)
as 160e-6 is for peak to peak amplitudes

psychoxiatao, 2018-12-02:
> the units of reject is always the unit of the channel type so for EEG it's Volt. To match what you did in EEGLAB it would be reject=dict(eeg=160e-6) as 160e-6 is for peak to peak amplitudes

Sorry, I still do not understand. 1 Volt = 1000000 uV,  160e-6=0.00016V=16uV right?  Why did it match(-80 uV, 80 uV) in EEGlab.

agramfort, 2018-12-02:
No it’s 160uV not 16 so twice 80uV