mne-tools/mne-python #5764: plot_joint passes wrong position information to plot_topomap

davidhaslacher, 2018-12-02:
Dear community,

I've just updated to 0.17 and noticed that calling AverageTFR.plot_joint() results in a ValueError when calling plot_topomap: 

ValueError("Data and pos need to be of same length. Got data of ""length %s, pos of length %s" % (len(data), len(pos)))

I believe this happens for the following reason: plot_joint() loads the position information for sensors from the layout file using find_layout() by default, which only excludes info['bads']. AverageTFR usually only contains a subset of the remaining channels. It seems that only position information for channels in info['ch_names'] should be passed to plot_topomap().

Best,

David

mmagnuski, 2018-12-02:
@jona-sassenhagen

davidhaslacher, 2018-12-02:
Yes, of course:

`import mne
from mne.datasets.brainstorm import bst_phantom_ctf
from mne.time_frequency import tfr_morlet
from mne.io import read_raw_ctf
import os.path as op
import sys
import numpy as np

sys.argv[0] = "--accept-brainstorm-license"
path = bst_phantom_ctf.data_path()
path = op.join(path, 'phantom_20uA_20150603_03.ds')
raw = read_raw_ctf(path,preload=True)
raw.pick_types(meg=True,ref_meg=True)
events = np.array([2000,3000,4000])
events = np.vstack((events, np.zeros_like(events), np.ones_like(events))).T
epochs = mne.Epochs(raw, events, tmin=-0.5, tmax=0.5)
freqs = [10]
n_cycles = [2]
power = tfr_morlet(epochs, freqs=freqs, n_cycles=n_cycles, return_itc=False)
timefreqs=[(0,10)]
power.plot_joint(timefreqs=None,title="Example TFR",show=True)`

This results in:

"ValueError: Data and pos need to be of same length. Got data of length 273, pos of length 275"

davidhaslacher, 2018-12-02:
Yes, of course:

```import mne
from mne.datasets.brainstorm import bst_phantom_ctf
from mne.time_frequency import tfr_morlet
from mne.io import read_raw_ctf
import os.path as op
import sys
import numpy as np

sys.argv[0] = "--accept-brainstorm-license"
path = bst_phantom_ctf.data_path()
path = op.join(path, 'phantom_20uA_20150603_03.ds')
raw = read_raw_ctf(path,preload=True)
raw.pick_types(meg=True,ref_meg=True)
events = np.array([2000,3000,4000])
events = np.vstack((events, np.zeros_like(events), np.ones_like(events))).T
epochs = mne.Epochs(raw, events, tmin=-0.5, tmax=0.5)
freqs = [10]
n_cycles = [2]
power = tfr_morlet(epochs, freqs=freqs, n_cycles=n_cycles, return_itc=False)
timefreqs=[(0,10)]
power.plot_joint(timefreqs=None,title="Example TFR",show=True)

```

This results in:

"ValueError: Data and pos need to be of same length. Got data of length 273, pos of length 275"

jona-sassenhagen, 2018-12-02:
Thanks, I will get on it - unless you want to contribute a fix yourself?

davidhaslacher, 2018-12-02:
Sure, I'll submit a pull request tomorrow.

jona-sassenhagen, 2018-12-02:
Great, thanks, be sure to @ me.