mne-tools/mne-python #5766: mne.viz.plot_evoked_topo(): all evoked times must be the same?

jjnurminen, 2018-12-03:
Howdy!

A student here tried to compare different datasets in mne.viz.plot_evoked_topo(), but the function bails due to slightly different sampling frequencies, giving the error that "all evoked times must be the same".

Should the plotter work even if the data are on different x grids, or is there a good reason for it not working?

larsoner, 2018-12-03:
> Should the plotter work even if the data are on different x grids, or is there a good reason for it not working?

It's likely just an implementation detail that it's easier internally to do the plotting if the `times` are all the same (e.g., choosing axis limits or ticks, combining data into arrays, etc.). So feel free to look to see if it can be generalized easily enough. I don't see any reason in principle that we can't or shouldn't support it, other than its implementation might be more complicated than it's worth.

jjnurminen, 2018-12-04:
Asking for identical time grids is a bit restrictive IMO, so it'd be nice to fix it.

If there's several plotting functions that assume identical times, it might be easier to write a gadget that interpolates a list of Evokeds into a common time grid? Then that could be called from any function.

agramfort, 2018-12-04:
I would rather see how to fix plotting functions

jjnurminen, 2018-12-04:
Of course matplotlib will interpolate the data anyway when plotting it, so interpolating it beforehand would be kind of superfluous...

PR #5788 (2019-01-12): MRG: handle different time vecs in plot_evoked_topo()
Closes #5766 
Since matplotlib can already overlay data on different x grids, this should an easy fix. Or maybe I'm missing something... let's see.
closed 2019-01-12 by agramfort