mne-tools/mne-python #5773: Broken link + bad rendering of plot_source_alighnment

massich, 2018-12-05:
[image]

this 3 bullets do not render properly in my computer + 1 `transformation matrices` points [here](https://mne-tools.github.io/stable/auto_tutorials/trans_matrices)

larsoner, 2018-12-05:
What's wrong with the bullet points?

massich, 2018-12-05:
Actually nothing. I just did not understood that actually for visualizing purposes the arrows have different lengths, (and maybe labels) that's why some letters are capital and surrounded by ( ). 

I thought it was a rendering issue.

massich, 2018-12-05:
maybe I would explain what RAS is, relate it to XYZ and then explain the arrows length. But I'm not sure if this is already done somewhere else.

larsoner, 2018-12-05:
That's what the bullet points are meant to do, feel free to reword if it's not clear

massich, 2018-12-05:
Actually, when I opened the issue, my goal was the broken link. And to know if we have some way to test that all links are right.

larsoner, 2018-12-05:
It should probably be:
```
`transformation matrices <trans_matrices_>`_
```
You can check to see if Sphinx provides a way to check that all links (including URL links inline) actually point somewhere valid, but doing so might slow things down a lot because it could cause potentially a *lot* of remote URL checks.

massich, 2018-12-06:
we can run it in an overnight build

GuillaumeFavelier, 2018-12-13:
Actually I found another broken link in [plot_introduction](https://martinos.org/mne/stable/auto_tutorials/plot_introduction.html?highlight=inverse) that points to https://martinos.org/mne/stable/auto_tutorials/auto_examples/index.html instead of https://martinos.org/mne/stable/auto_examples/index.html

[image]

PR #5791 (2018-12-17): MRG: Check links
This PR:

1. Modernizes CircleCI with a workflow and nice names.
2. Adds a ~nightly~ weekly cron job to check links.
3. Fixes all our broken links except an `IPython` one (as PoC).

Once CircleCI worksand tells us properly that things are broken, I'll:

- [x] fix the last broken link
- [x] remove `linkcheck` step from the `default` workflow

Closes #5773.
closed 2018-12-17 by agramfort