mne-tools/mne-python #5778: Installation instructions do not work on Mac OSX

noahbenson, 2018-12-06:
I recently had to reinstall MNE on Mac OSX (Mojave), and to do so I followed the [installation instructions](https://www.martinos.org/mne/stable/install_mne_python.html) exactly. They appeared to work, but when I started testing out the installation it became clear that the graphics are not working. 

The problem appears to be with mayavi/pyqt/vtk, and similar bug reports have been documented [in](https://stackoverflow.com/questions/14517443/enthought-mayavi-animation-blank-display) [many](https://github.com/enthought/mayavi/issues/532) [projects](https://github.com/enthought/mayavi/issues/257), including [mne](https://github.com/mne-tools/mne-python/issues/760). None of the proposed solutions have made any difference as far as I can tell, however.

## Info and Reproduction:

```
(mne) $ ipython
Python 3.6.6 | packaged by conda-forge | (default, Jul 26 2018, 09:55:02) 
Type 'copyright', 'credits' or 'license' for more information
IPython 6.5.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import mne

In [2]: mne.sys_info()
Platform:      Darwin-18.0.0-x86_64-i386-64bit
Python:        3.6.6 | packaged by conda-forge | (default, Jul 26 2018, 09:55:02)  [GCC 4.2.1 Compatible Apple LLVM 6.1.0 (clang-602.0.53)]
Executable:    /Users/nben/Library/miniconda2/envs/mne/bin/python
CPU:           i386: 8 cores
Memory:        48.0 GB

mne:           0.17.0
numpy:         1.15.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.1.0
matplotlib:    2.2.3 {backend=MacOSX}

sklearn:       0.19.1
nibabel:       2.3.0
mayavi:        4.7.0.dev0 {qt_api=pyqt5, PyQt5=5.9.2}
cupy:          Not found
pandas:        0.23.4
dipy:          0.14.0

In [3]: from mayavi import mlab

In [4]: %gui qt

In [5]: mlab.test_plot3d()
Out[5]: <mayavi.modules.surface.Surface at 0x135f0d620>
```
[image]

(The visualization window simply hangs.)

Is there something obviously wrong with my configuration? Has anyone managed to make this work in recent versions of Mac OSX?

larsoner, 2018-12-06:
Does it "un-hang" if you do:
```
>>> mlab.process_ui_events()
```
If so then it's just a problem with the GUI event loop not running, you might need `%gui qt`.

If that works, we should probably change the note about this in the installation instructions to be non-Windows specific. Can you open a PR to do this?

larsoner, 2018-12-06:
.. oh you have the `% gui qt` in there. Weird. Let me know how the `process_ui_events` goes and if it works, we can look into why it's not actually starting the event loop.

You can check to see if it is by doing:
```
import matplotlib.pyplot as plt
fig, axes = plt.subplots()
```
and seeing if that window is responsive.

noahbenson, 2018-12-06:
Yeah, sorry, I followed them initially then tried to fix things myself by upgrading/installing/downgrading various packages based on other answers, then wrote the post. The same thing happens in just about every scenario (except in some where the Python kernel outright crashes on `from mayavi import mlab`).

That said, I just cleared my mne environment and reran the install instructions--same problem. But I noticed a warning about my sha256 hashes not matching with the Mayavi package during install, so I deleted the cache files and reran the install instructions and now things are working fine. So I'm going to guess that this had nothing to do with MNE :-)

Thanks!

closed 2018-12-06 by noahbenson

larsoner, 2018-12-06:
Weird -- yes, sounds like a `pip` caching problem. Maybe we should add a note to the troubleshooting guide about telling `pip` not to use cache (or deleting the cache) if things don't work?