mne-tools/mne-python #5779: Visual C++ (still) required when installing?

jjnurminen, 2018-12-07:
On Windows 7, installing with `conda env create -f environment.yml` (environment from latest master) resulted in:

    error: Microsoft Visual C++ 14.0 is required. Get it with "Microsoft Visual C++ Build Tools": 
    https://visualstudio.microsoft.com/downloads/

larsoner, 2018-12-07:
In theory no it should not be required. Perhaps `dipy` has a compiler dependency, or `mayavi` still isn't quite working right. Can you tell from the traceback which library requires it?

jjnurminen, 2018-12-10:
Sorry for the overly terse report. Seems to be dipy and nitime (whatever they are):

    running build_ext
    error: Microsoft Visual C++ 14.0 is required. Get it with "Microsoft Visual C++ Build Tools": 
    https://visualstudio.microsoft.com/downloads/

     ----------------------------------------
     Failed building wheel for dipy

...

    running build_ext
    skipping 'nitime\_utils.c' Cython extension (up-to-date)
    building 'nitime._utils' extension
    error: Microsoft Visual C++ 14.0 is required. Get it with "Microsoft Visual C++ Build Tools": 
    https://visualstudio.microsoft.com/downloads/

    ----------------------------------------
    Failed building wheel for nitime

mayavi seems to build ok

jjnurminen, 2018-12-11:
`conda install cython` does not seem to install any C compiler package.

pip seems to have a --binary-only flag, would it be possible to use this and avoid building packages from source? I tried to add the flag to the pip section of environment.yml (which apparently should be supported), but it caused a conda internal error.

In desperation I even tried to install MS Visual Studio, but that also went wrong (the installer somehow fails to download  files).

jjnurminen, 2018-12-11:
I wouldn't mind installing gcc, but MS Visual Studio as a requirement for mne is a bit too much :)

Installing dipy in my existing Conda env (pip install dipy) works fine and does not attempt to build anything from source. However for some reason, pip tries to build dipy from source when installing via environment.yml. Maybe the environment pulls a specific version of dipy that does not have binary builds available?

jjnurminen, 2018-12-11:
As for testing-related deps, maybe they could be (only) included in another file such as `environment_dev.yml`? People who need to run tests could use that one to create their environment.

jjnurminen, 2018-12-11:
Ok, I managed to install by deleting the dipy and nitime lines from `environment.yml`, and then running

    pip install --only-binary dipy dipy

in the newly created environment. So probably this can be solved by version pinning or pip options.

larsoner, 2018-12-11:
@jjnurminen can you see if https://github.com/mne-tools/mne-python/pull/5787 works for you? It's probably the same solution you found, namely putting `--only-binary dipy` in the `dipy` requirement line in `environment.yml`

jjnurminen, 2018-12-12:
#5787 indeed seems to work, i.e. I'm able to create the environment without errors and run `python -c import mne`

agramfort, 2018-12-12:
pull-request?

PR #5787 (2018-12-12): MRG: --only-binary dipy
Closes #5779.

First commit just to see if we can replicate #5779. After that, fix it.

Replicated: https://ci.appveyor.com/project/mne-tools/mne-python/builds/20931971
closed 2018-12-12 by agramfort