mne-tools/mne-python #5781: ICA on MEG reference channels?

jshanna100, 2018-12-10:
1) What is the reason why MEG reference channels are not allowed to be submitted to an ICA fit? 2) If there is not a compelling reason, what would be the best procedure for changing the channel type from meg reference to meg? As far as I can tell channels.set_channel_types cannot do this.

jshanna100, 2018-12-10:
Yes. If you attempt to pass in meg reg channels, they'll just get caught by _check_for_unsupported_ica_channels in preprocessing.ica, line 116.

agramfort, 2018-12-12:
I guess we need a PR and a fix to allow ref_meg in
_check_for_unsupported_ica_channels

do you have time to open a PR ?

jshanna100, 2018-12-13:
Will do latest end of next week. I've hacked out a temporary solution, and am investigating whether ICA decompositions of these channels can actually be useful in noise removal, and if existing visualisation functions can be applied to them with only small modifications.

Of course, fixing _check_for_unsupported_ica_channels is only the beginning. Problems continue to appear downstream, and they would all need to be fixed as well.

jshanna100, 2018-12-19:
#5807

closed 2019-03-14 by jshanna100