mne-tools/mne-python #5784: time-frequency beamformer on single trial

romquentin, 2018-12-10:
Hi all, 

I am trying to do MVPA decoding from time-frequency transform epoch data (power) in source space.

The first thing I tried is, schematically:

1. `raw = mne.Epochs() #  Create epochs`
1. `stcs = apply_lcmv_epochs(epochs, filters)  # Source transform at the single trial level  `
1. `for jj, stc in enumerate(stcs):  # Rearange the data`
    `         X_data[jj] = stc.data`
1. `X = tfr_array_morlet(X_data, output=power)  # Compute single trial time-frequency in source`
1. `clf = SlidingEstimator(make_pipeline(StandardScaler(), LinearModel(Ridge()))) #  Create regressor`
1. `clf.fit(X[train], y[train]) and clf.score(X[test], y[test]) + get_coef('patterns_')  # Run regressor and get the pattern at each cv iteration`

For now, it is crashing because it exceeds memory limit on my server (the X array in line 4 is big)
It doesn't seem to be the most efficient way to do that. Do you have any advices?
I was wondering if it exists an equivalent of `mne.beamformer.tf_lcmv` at the single trial level?

Thank you for your help!

Romain

agramfort, 2018-12-11:
what are you aiming to show? estimate? Maybe there is a simpler route.

mmagnuski, 2018-12-16:
@romquentin Do you use `decim` when using time-frequency? You could also do the analysis in parts if the full approach eats too much of your memory (chunk time-frequency space and process each chunk independently and then glue the results).