mne-tools/mne-python #5789: default evoked.plot to spatial_colors=True?

jona-sassenhagen, 2018-12-13:
I sometimes see people use `evoked.plot_joint` and manually crop the topoplots. This probably is because they don't check the kwargs for evoked.plot ... Does anything speak against defaulting `evoked.plot` to `spatial_colors=True` at this time?

larsoner, 2018-12-13:
I suspect `evoked.plot` is one of the most widely used functions, with lots of existing code that use it. So we should not change its defaults without good reason. I'm not sure if moving to `spatial_colors=True` is worth it here.

Since `plot_joint` is probably less commonly used, and passes kwargs to `evoked.plot` effectively, we could change the default *there* to be `ts_kwargs=None` means `ts_kwargs=dict(spatial_colors=True)`.

I'm not sure what this spatial colors stuff has to do with cropping topoplots, though.

jona-sassenhagen, 2018-12-13:
`we could change the default there to be ts_kwargs=None means ts_kwargs=dict(spatial_colors=True).`

I'm not sure I get what you mean - `plot_joint` defaults to `spatial_colors=True`.

`I'm not sure what this spatial colors stuff has to do with cropping topoplots, though.`

People who want spatial colors use `plot_joint` and crop the figure to get something that looks like `plot(spatial_colors=True)`.

larsoner, 2018-12-13:
Ahh I see what you mean. They vertically chop it out.

I'm -0.5 changing the default on this, to me it doesn't seem worth the large volume of code breakages / warnings people will get. But I'm fine with it if it's a very prevalent problem or others see this sort of thing, too

mmagnuski, 2018-12-13:
I'd be +0.5 on changing the default to spatial colors. People seem to like it (I do too), it actually adds valuable information in very esthetical form and this would make `plot()` and `plot_joint()`  look more similar. Or - we could also add `chop_out_topos` to `plot_joint` ;)

mmagnuski, 2018-12-13:
@agramfort This magic is already present in plot_joint so it's a kind of magic that some users might have grown accustomed to. ;) But making `spatial_colors=True` more visible for `.plot()` docs is ok too.

mmagnuski, 2018-12-13:
(unless plot_joint does not work without channel pos - due to the topomaps, in that case I agree)

agramfort, 2018-12-13:
I think you need to think about the user base as a whole here.

as @larsoner said evoked.plot() has been in black since day one.
It's the default behavior for all our users now. It's not because
one user ask for it that you should impose this to the majority.

It's not a bug fix but a matter of taste. If you want to have the
opinion of user base make a poll on the mailing list. If majority
votes for it we should consider it.

mmagnuski, 2018-12-13:
> It's not a bug fix but a matter of taste. If you want to have the
opinion of user base make a poll on the mailing list. If majority
votes for it we should consider it.

Maybe @jona-sassenhagen would like to conduct the poll, I got convinced by you and @larsoner :)

jona-sassenhagen, 2018-12-20:
Not another poll :D

closed 2018-12-20 by jona-sassenhagen