mne-tools/mne-python #5792: issue with running mne on 32bit OS

monkeyman192, 2018-12-14:
#### Describe the bug
So this will be a pretty niche issue I think as most people (rightfully so) run 64 bit versions of everything these days, however, due to various reasons we have to be using a 32bit computer in our lab.
Because of this I found an issue in `mne.io.utils:_mult_cal_one`. On a 32 bit machine a `MemoryError` is raised by numpy.

#### Steps and/or code to reproduce
I was running `mne-bids` to do BIDS conversion on the computer and it was all fine up until the extraction of event data at which point the aforementioned `MemoryError` occurred.
https://github.com/mne-tools/mne-python/blob/master/mne/io/utils.py#L93
is the specific line at which it breaks.

#### Additional information
I was able to fix this issue by replacing the line with:
```python
try:
    np.take(one, idx, axis=0, out=data_view)
except MemoryError:
    one = one[idx]
```

I can make a PR if this is a satisfactory solution, but I don't know if there are other places this issue could occur. I tried to debug a little bit more to determine which array was causing the `MemoryError` but it was a bit hard with how I was running it to do that.
Further, this shouldn't even be a big array (there was like, one event...)

agramfort, 2018-12-14:
is it a dtype issue? I don't think the try except is the right fix. np.take
with out param
should not lead to memory allocation issues. If it's actually a dtype issue
we should
just enforce the right dtypes.

monkeyman192, 2018-12-18:
Ok, so after having a bit more of an investigation I can say that the issue is probably with the dtype.
The dtype is `float64` which I assume causes the issue.
What I don't really understand is why this only happens with the `take` function, and not with any of the other times (for example with the line)
```python
one = np.asarray(one, dtype=data_view.dtype)
```
Although, I just checked the sizes of `one` and `dataview` by using `numpy.ndarray.nbytes` with the following results for my data:
```
data_view.nbytes = 1562496
one.nbytes = 401561472
```
The value for `one` seems very high, especially since it should be pretty much empty...

agramfort, 2018-12-18:
sorry what is data_view?

if you do:

`np.take(one, idx, axis=0, out=data_view)`

then one should have:

data_view.dtype == one.dtype
data_view.shape == (len(idx), one.shape[1])

is it the case?

monkeyman192, 2018-12-18:
I don't know what `data_view` is. I'm just looking through the code that is failing.

`np.take(one, idx, axis=0, out=data_view)` is the line that has a `MemoryError`, but the following two equalities are true from what I can see by printing out values.

larsoner, 2019-08-21:
I'm going to close this as a corner case out-of-memory bug that we probably don't need to fix, and might have been / should be fixed in `numpy`

closed 2019-08-21 by larsoner