mne-tools/mne-python #5794: epochs realignment: shifting time by variable length on each epoch.

kdoelling1919, 2018-12-15:
#### Describe the problem
I have a dataset where subjects listen to a series of 4-5 tones on each trial. However, the Inter Onset Interval between tones is different from trial to trial. I would like to create a function that allows me to realign epochs so that the 0 in `epoch.times` is now centered on the 2nd, third, final, etc tone. Typically this is done by altering the event array directly. However, because of the series of tones, I still want my baseline to always be before the first tone where there is silence.
So I want to be able to:
1) define large epochs that cover all tones in each trial
2) Apply a baseline, timelocked to pre first tone, in an epoch by epoch manner
3) Realign epochs to a specific tone in the trial. Critically, this shift would be by a __different amount for each epoch so `epochs.shift_time` wouldn't work.__

#### Describe your solution
I have found a solution to do this outside of mne and would like to implement it in a TimeMixin so that it can be used for both Epochs, EpochsTFR and wherever else there are multiple trials with time stored, (Source Estimates?). 

It's a bit ugly and hard to follow, which is exactly why I think we should implement it so that it's well tested and users can avoid the potential errors when they try to do it outside of mne. Here is a very basic example which shows the concept of the method.
```
import numpy as np
t = np.round(np.linspace(-.5, 1., 16), 1)
sfreq = 1. / (t[1] - t[0])
n_epochs = 3
# faux data, 3 epochs x 16 timepoints
data = t * np.ones(n_epochs)[:, None]

# How many samples to shift for each epoch in realignment
sampshift = np.array([2, 4, 6])

# realigned epochs must have smaller window than original epoch
new_tmin, new_tmax = -.2, .3

# indices to extract the correct realigned samples
minsamp = np.round((new_tmin - t[0]) * sfreq).astype(int)
maxsamp = np.round((new_tmax - new_tmin) * sfreq).astype(int) + minsamp
idx = sampshift[:, None] + np.arange(minsamp, maxsamp + 1)

# needs some checks here to make sure you don't exceed epoch size
shift_data = data[np.arange(n_epochs)[:, None], idx]
```

I would also like to move some methods like `.shift_time,` maybe `.crop` into the TimeMixin so that they are not implemented in multiple places. This could be a separate PR.

#### Describe possible alternatives
Another alternative for my specific problem is to allow for different baseline windows for each epoch but this doesn't seem to me to be a good practice generally and I'm not sure how much we want to encourage people to draw baselines from different timepoints in a typical use case.

agramfort, 2018-12-16:
what API do you suggest? indeed it's not easy to do presently.

kdoelling1919, 2018-12-16:
I'm not 100% sure this answers your question, but I was initially thinking of adding it as a method `epochs.realign()` to the class to operate on epochs in place. But I could also see it as a function `realigned_epochs = mne.realign_epochs(epochs)`. This would work well for me as there are multiple points in the trial I would like to realign to so it might be good to generate new epochs rather than altering the original.

agramfort, 2018-12-16:
how would you specify offsets applied?

larsoner, 2018-12-17:
This seems like the sort of manipulation that is specific enough to justify using custom NumPy operations and EpochsArray rather than adding a function to MNE. There are potentially lots of custom baseline, re-epoching, sub-epoching, custom rejection, etc. use cases users can have, and rather than to to support them in MNE it's more sustainable to make sure it's clear how to leverage NumPy in combination with MNE.

kdoelling1919, 2018-12-17:
Hmmm, I understand the danger of trying to include every possible custom feature in MNE. My feeling, though, is that the need to realign epochs is becoming more common as experiments with continuous stimuli are becoming in vogue. For example, I have another experiment where participants listen to clips of music for which I have the timing of each note and its identity. I can think of a few other projects in our lab that would also benefit from this for data ranging from sentences/stories to classroom interactions. The need to identify a clean baseline outside the continuous stream followed by re-epoching within seems to me more and more useful. This is quite simple in FieldTrip using [ft_redefinetrial](https://github.com/fieldtrip/fieldtrip/blob/master/ft_redefinetrial.m) but not so in MNE. 

I could definitely do it with NumPy and EpochsArray but as I mentioned since it's not super obvious how to do this and getting it wrong could be *very* detrimental, I think it would be a good idea to have it implemented as a well-tested function or method. Definitely willing to be wrong though if others disagree!

jona-sassenhagen, 2018-12-17:
Isn't this also a bit like EEGLAB's realignment? 

[image]

agramfort, 2018-12-17:
one thing that worries me is the risk of errors due to dropping of bad
epochs. If realign
function takes a plain array how do you make sure it applies to
epochs.selection etc.

kdoelling1919, 2018-12-17:
I can imagine a couple of solutions to this.

1. Allow for some flexibility in array size, allowing for it to be either the length of the current number of epochs or the length of the original events used to initialized epochs. Something like this
```
if len(offset) == len(epochs.drop_log):
    offset = offset[epochs.selection]
elif len(offset) != len(epochs):
    raise ValueError('offset has incorrect length')
```

2. Require that offset be a column in metadata and instead of asking for the numpy array, it asks for the name of the column as a `str`.

1 is based only on the minimum dependencies but maybe is overly flexible?
2 definitely gets rid of the potential for dropping bad epochs but requires pandas

Probably, if option 1 is okay, then we should allow for both: if input is an array, do 1; if string, do 2. So really the question is whether option 1 is a reasonable solution.

agramfort, 2018-12-17:
1 is dangerous / magic.

2 should not require pandas. You can have metadata without pandas as
dict of arrays.

I am not opposed to discussion but I would suggest to add with such a feature
a dataset that demonstrate why and when to do something like this. Can you think
of some public data we could use?

larsoner, 2018-12-17:
We have an example somewhere plots/reorders things by RT. That could be adapted to re-epoch based on RT?

jona-sassenhagen, 2019-03-24:
>We have an example somewhere plots/reorders things by RT. That could be adapted to re-epoch based on RT?

https://mne-tools.github.io/dev/auto_examples/visualization/plot_roi_erpimage_by_rt.html#sphx-glr-auto-examples-visualization-plot-roi-erpimage-by-rt-py