mne-tools/mne-python #5802: Temporal connectivity metrics

brovelli, 2018-12-18:
Dear all,

at the lab, we are developing a set of functions for directed functional connectivity analysis in the time-domain. These measures are complementary to the frequency-domain metrics, which are already present in mne.connectivity.spectral_connectivity

The list of metrics will not be exhaustive, but we would like to have simple measures to be accessible from MNE. These may include covariance-based Granger causality, Transfer Entropy. 

Has anyone already started a similar project in MNE or digging into this issue?

What is your feeling about a similar project? Would a mne.connectivity.temporal_connectivity function be worth the effort?

Thanks for your comments and help

Andrea

agramfort, 2018-12-18:
I am not aware of such an effort.

I would recommend to start from a separate project to allow to iterate
quickly.

You would expose functions that take mne objects as inputs?

brovelli, 2018-12-18:
@cbrnr : Thanks for the link, we'll look into it. By time-domain FC measures I mean those do not have a frequency-specific decomposition such as Coherence. I normally use the notation described in [Bastos et al. 2016](https://www.frontiersin.org/articles/10.3389/fnsys.2015.00175/full)

@agramfort : well, the long-term idea would be to make this code compatible with mne objects. We'll probably start off from a separate project.

Thanks again and do not hesitate to suggest coding strategies etc.