mne-tools/mne-python #5806: Test suite fails when trying to build Debian package

tillea, 2018-12-19:
Hi,
I tried to work on the Debian packaging which was lagging behind upstream heavily.  It seems the matplotlib [issue](https://bugs.debian.org/896025) is solved but when I try to build version 0.17 I've got

    ...
    make[1]: Entering directory '/build/python-mne-0.17+dfsg'
    mkdir -p build
    HOME=/build/python-mne-0.17+dfsg/build \
    MNE_DONTWRITE_HOME=true MNE_SKIP_SAMPLE_DATASET_TESTS=true MNE_FORCE_SERIAL=true MNE_SKIP_NETWORK_TESTS=1 \
      xvfb-run --auto-servernum --server-num=20 -s "-screen 0 1024x768x24 -ac +extension GLX +render -noreset" \
      py.test -s -v mne
    ============================= test session starts ==============================
    platform linux2 -- Python 2.7.15+, pytest-3.10.1, py-1.7.0, pluggy-0.8.0 -- /usr/bin/python2
    cachedir: .pytest_cache
    rootdir: /build/python-mne-0.17+dfsg, inifile: setup.cfg
    plugins: cov-2.6.0
    collecting ... Using default location ~/mne_data for testing...
    Creating ~/mne_data
    Using default location ~/mne_data for testing...
    Using default location ~/mne_data for testing...
    ...
    Using default location ~/mne_data for testing...
    Using default location ~/mne_data for testing...
    Segmentation fault
    make[1]: *** [debian/rules:26: override_dh_auto_test] Error 139
    make[1]: Leaving directory '/build/python-mne-0.17+dfsg'

Kind regards, Andreas.

agramfort, 2018-12-19:
thanks @tillea for taking a stab at this.

what would be an easy way to replicate this issue? We would need to identify which test breaks.

larsoner, 2018-12-19:
It would help to install `python-faulthandler`, it should show where it breaks. I'd bet it's related to `mayavi` / `vtk`.

larsoner, 2018-12-19:
`pytest-faulthandler`, rather

tillea, 2018-12-19:
On Wed, Dec 19, 2018 at 04:45:24AM -0800, Alexandre Gramfort wrote:
> thanks @tillea for taking a stab at this.
> 
> what would be an easy way to replicate this issue? We would need to identify which test breaks.

Well, your are listed yourself as "Uploader" in the Debian package. ;-)
You can reproduce it easily by building the package in short

    $ gbp clone https://salsa.debian.org/med-team/python-mne
    $ cd python-mne
    $ gbp build-package

(may be after adding python-faulthandler to Build-Depends).
If you have no idea what I'm talking about its probably better if I remove your ID from the list of Uploaders.  Howeder, if you really want to stick to this role which would be really appreciated I'd happily give some more hints if needed.
Kind regards, Andreas.

tillea, 2018-12-19:
On Wed, Dec 19, 2018 at 04:53:12AM -0800, Eric Larson wrote:
> `pytest-faulthandler`, rather
In the build scenario the error is produced is a Debian schroot which is automatically created.  Since pytest-faulthandler is not packaged for Debian this will not work that easily.
Thanks for the hint anyway, Andreas.

tillea, 2019-01-11:
Ping?

larsoner, 2019-01-11:
In case @agramfort does not have time now to work on fixing the package (and I suspect he does not), I can look into maintaining the `python-mne` package in Debian. I know MNE's requirements and how to debug, but I have never maintained a package for Debain. Should I just search for tutorials / documentation, and start from there?

tillea, 2019-01-11:
@larsoner May be you can even reproduce (and fix) the reported issue without dealing with the packaging?  For sure its really welcome if someone would actively care for the package.  The [Debian Med team policy](https://med-team.pages.debian.net/policy/#readings) lists some interesting reading and we even have a [Mentoring program](https://wiki.debian.org/DebianMed/MoM) in our team.  But may be the first step would be to reproduce the issue above.

larsoner, 2019-01-11:
The segfault is likely related to some other package (probably python-mayavi), because ours is pure Python. None of our continuous integration services (nor me locally) reproduce it. So it probably also has to do with the particular setup of the machine doing the Debian build, including it's graphics setup (e.g., if an X server is running and if so, what its OpenGL capabilities are).

Hence why I'm thinking it's necessary for whoever debugs this to have a setup equivalent to that of the Debian builder that experiences the error...

tillea, 2019-01-11:
On Fri, Jan 11, 2019 at 12:15:36PM -0800, Eric Larson wrote:
> The segfault is likely related to some other package (probably python-mayavi), because ours is pure Python. None of our continuous integration services (nor me locally) reproduce it. So it probably [...]

Debian packages are usually built in a chroot which is *disconnected* from the internet and has no X server running.  It is possible to starte a fake X server by

    xvfb-run --auto-servernum --server-args="-screen 0 1024x768x24"

but if it comes to OpenGL its not really predictable whether the build will be successful.  So if you are *sure* that the very test is related to a X server issue deactivating this test for the Debian build would be the most straightforward solution.
Thanks a lot for your input, Andreas.

tillea, 2019-01-14:
I've checked the packaging and the test is actually using [xvfb-run](https://salsa.debian.org/med-team/python-mne/blob/master/debian/rules#L28).  So I think its a possible course of action to exclude the test that causes SIGSEGV - but which one is it???

larsoner, 2019-01-14:
There are a bunch of them, basically anything that has the `requires_mayavi` decorator.

But really we should fix the problem. #5831 probably does. Feel free to try that `master` branch just to see if it does. In the meantime I'll see if I can get a proper debian build testing environment set up.

tillea, 2019-01-15:
On Mon, Jan 14, 2019 at 06:44:15AM -0800, Eric Larson wrote:
> There are a bunch of them, basically anything that has the `requires_mayavi` decorator.
> 
> But really we should fix the problem. #5831 probably does. Feel free to try that `master` branch just to see if it does. In the meantime I'll see if I can get a proper debian build testing environment set up.

Please excuse my ignorance but what actual patch should I try.  Please note: I'm absolutely not involved in mne (do not even have any idea what it is good for).  I simply try to care for the package since nobody else finds some spare time for it.  So pointing me to a patch I could apply would help a lot to speed things up.  Kind regards, Andreas.

larsoner, 2019-01-15:
Can you switch to the latest `master` branch easily? If not I'll probably need to make a patch specific to `maint/0.17`

tillea, 2019-01-15:
On Tue, Jan 15, 2019 at 07:37:58AM -0800, Eric Larson wrote:
> Can you switch to the latest `master` branch easily? If not I'll probably need to make a patch specific to `maint/0.17`

Hmmmm, I could but what else do I get?  The result of this will make it (hopefully!) into the next Debian stable release.  I do not really like a "random development state".  It would be more convenient if you would do a point release if you are happy with the current state.  Kind regards, Andreas.

larsoner, 2019-01-15:
> I do not really like a "random development state".  It would be more convenient if you would do a point release if you are happy with the current state.

Agreed. We will do that once we isolate the problem. My idea of running latest `master` / random development state is just to give us more information about if the fix I put in `master` in #5831 has a decent chance of working in the Debian build. Of the changes we have pushed since releasing 0.17, this change is by far the most likely thing to fix a `segfault`, so it would be informative (especially if it still fails!).

I'm currently working on porting it to `maint/0.17` in #5837. You could test `master` in the meantime if it's easy. If it's not, I hope to get #5837 in today or tomorrow, and you could test that instead. (If it works, it's probably what we would push to `pypi` as 0.17.1.)

larsoner, 2019-01-15:
@tillea can you try the `maint/0.17` branch? If it works we'll cut a `0.17.1` release ASAP.

tillea, 2019-01-16:
No change for maint/0.17.  I attached the full build log:
[python-mne_0.17.1-1_amd64.build.zip](https://github.com/mne-tools/mne-python/files/2763175/python-mne_0.17.1-1_amd64.build.zip)
I'm somehow tempted to ignore build time test errors in general since we really should get the Debian package out and I can not spent hours on this task which is very low on my priority list.

larsoner, 2019-01-17:
I can get it to crash on Ubuntu 18.10 with:
```
$ gbp clone https://salsa.debian.org/med-team/python-mne
$ cd python-mne
$ sudo apt-get install python-all python-nibabel python-numpy python-scipy python-sphinx python-nose python-pytest python-pytest-cov python-matplotlib python-joblib python-sklearn mayavi2 yui-compressor
$ gbp build-package --git-ignore-new
```
I tracked it down on my system at least to a problem segfault with wxPython. But I agree it's probably not trivial to get it fixed. +1 for ignoring the test time errors @tillea. Alternatively, I can make it so that mayavi and/or GUI tests are skipped. Let me know what you prefer -- whatever's fastest for you!

Also I noticed some pending changesets (?), some of which we could easily incorporate in MNE-Python rather than having them be build-specific.

FYI for future reference this is what I can get:
```
larsoner@bunk:~/python/mne-python$ ~/.local/bin/pytest mne/gui/
=============================================================================== test session starts ================================================================================
platform linux2 -- Python 2.7.15+, pytest-4.1.1, py-1.7.0, pluggy-0.8.1
rootdir: /home/larsoner/python/mne-python, inifile: setup.cfg
plugins: faulthandler-1.5.0, cov-2.5.1
collected 13 items                                                                                                                                                                 

mne/gui/tests/test_coreg_gui.py ..Fatal Python error: Segmentation fault

Thread 0x00007f6e01e24700 <pytest> (most recent call first):
  File "/usr/lib/python2.7/dist-packages/traitsui/image/image.py", line 292 in _process
  File "/usr/lib/python2.7/threading.py", line 754 in run
  File "/usr/lib/python2.7/threading.py", line 801 in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 774 in __bootstrap

Thread 0x00007f6e02625700 <pytest> (most recent call first):
  File "/usr/lib/python2.7/dist-packages/traitsui/image/image.py", line 292 in _process
  File "/usr/lib/python2.7/threading.py", line 754 in run
  File "/usr/lib/python2.7/threading.py", line 801 in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 774 in __bootstrap

Thread 0x00007f6e03026700 <pytest> (most recent call first):
  File "/usr/lib/python2.7/threading.py", line 340 in wait
  File "/usr/lib/python2.7/Queue.py", line 168 in get
  File "/home/larsoner/python/mne-python/mne/gui/_coreg_gui.py", line 1345 in worker
  File "/usr/lib/python2.7/threading.py", line 754 in run
  File "/usr/lib/python2.7/threading.py", line 801 in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 774 in __bootstrap

Thread 0x00007f6e03aa7700 <pytest> (most recent call first):
  File "/usr/lib/python2.7/threading.py", line 340 in wait
  File "/usr/lib/python2.7/Queue.py", line 168 in get
  File "/home/larsoner/python/mne-python/mne/gui/_coreg_gui.py", line 1345 in worker
  File "/usr/lib/python2.7/threading.py", line 754 in run
  File "/usr/lib/python2.7/threading.py", line 801 in __bootstrap_inner
  File "/usr/lib/python2.7/threading.py", line 774 in __bootstrap

Current thread 0x00007f6e3f6ef740 <pytest> (most recent call first):
  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk3/wx/_core.py", line 11176 in GetHandle
  File "/usr/lib/python2.7/dist-packages/vtk/wx/wxVTKRenderWindowInteractor.py", line 627 in Render
  File "/usr/lib/python2.7/dist-packages/vtk/wx/wxVTKRenderWindowInteractor.py", line 420 in OnSize
  File "/usr/lib/python2.7/dist-packages/tvtk/pyface/ui/wx/scene.py", line 537 in OnSize
  File "/usr/lib/python2.7/dist-packages/wx-3.0-gtk3/wx/_core.py", line 14838 in Layout
  File "/usr/lib/python2.7/dist-packages/tvtk/pyface/ui/wx/decorated_scene.py", line 107 in _create_control
  File "/usr/lib/python2.7/dist-packages/tvtk/pyface/tvtk_scene.py", line 209 in __init__
  File "/usr/lib/python2.7/dist-packages/tvtk/pyface/ui/wx/scene.py", line 285 in __init__
  File "/usr/lib/python2.7/dist-packages/tvtk/pyface/ui/wx/decorated_scene.py", line 69 in __init__
  File "/usr/lib/python2.7/dist-packages/tvtk/pyface/ui/wx/scene_editor.py", line 100 in _create_scene
  File "/usr/lib/python2.7/dist-packages/tvtk/pyface/ui/wx/scene_editor.py", line 59 in init
  File "/usr/lib/python2.7/dist-packages/traitsui/editor.py", line 156 in prepare
  File "/usr/lib/python2.7/dist-packages/traitsui/wx/ui_panel.py", line 1031 in add_items
  File "/usr/lib/python2.7/dist-packages/traitsui/wx/ui_panel.py", line 740 in add_dock_window_splitter_item
  File "/usr/lib/python2.7/dist-packages/traitsui/wx/ui_panel.py", line 700 in add_dock_window_splitter_items
  File "/usr/lib/python2.7/dist-packages/traitsui/wx/ui_panel.py", line 652 in __init__
  File "/usr/lib/python2.7/dist-packages/traitsui/wx/ui_panel.py", line 523 in fill_panel_for_group
  File "/usr/lib/python2.7/dist-packages/traitsui/wx/ui_panel.py", line 349 in panel
  File "/usr/lib/python2.7/dist-packages/traitsui/wx/ui_live.py", line 223 in init
  File "/usr/lib/python2.7/dist-packages/traitsui/wx/ui_live.py", line 99 in ui_dialog
  File "/usr/lib/python2.7/dist-packages/traitsui/wx/ui_live.py", line 68 in ui_live
  File "/usr/lib/python2.7/dist-packages/traitsui/wx/toolkit.py", line 140 in ui_live
  File "/usr/lib/python2.7/dist-packages/traitsui/ui.py", line 221 in ui
  File "/usr/lib/python2.7/dist-packages/traitsui/view.py", line 433 in ui
  File "/usr/lib/python2.7/dist-packages/traits/has_traits.py", line 1871 in edit_traits
  File "/home/larsoner/python/mne-python/mne/gui/__init__.py", line 16 in _initialize_gui
  File "/home/larsoner/python/mne-python/mne/gui/__init__.py", line 171 in coregistration
...
```

tillea, 2019-01-18:
On Thu, Jan 17, 2019 at 02:08:34PM -0800, Eric Larson wrote:
> I can get it to crash on Ubuntu 18.10 with:
> ```
> $ gbp clone https://salsa.debian.org/med-team/python-mne
> $ cd python-mne
> $ sudo apt-get install python-all python-nibabel python-numpy python-scipy [...]

Good to know.

> But I agree it's probably not trivial to get it fixed. +1 for ignoring the test time errors @tillea. Alternatively, I can make it so that mayavi and/or GUI tests are skipped. Let me know what you prefer -- whatever's fastest for you!

The fastest for me is

    diff --git a/debian/rules b/debian/rules
    index 991c18035..037579695 100755
    --- a/debian/rules
    +++ b/debian/rules
    @@ -26,7 +26,9 @@ override_dh_auto_test:
            HOME=$(CURDIR)/build \
            MNE_DONTWRITE_HOME=true MNE_SKIP_SAMPLE_DATASET_TESTS=true MNE_FORCE_SERIAL=true MNE_SKIP_NETWORK_TESTS=1 \
              xvfb-run --auto-servernum --server-num=20 -s "-screen 0 1024x768x24 -ac +extension GLX +render -noreset" \
    -         $(TESTRUNNER) -s -v mne
    +         $(TESTRUNNER) -s -v mne || true
    +       # ignore remaining errors in test suite
    +       # see https://github.com/mne-tools/mne-python/issues/5806

 
> Also I noticed some pending changesets (?), some of which we could easily incorporate in MNE-Python rather than having them be build-specific.

Feel free to check the patches we applied and incorporate whatever is not too Debian specific.  The duty of the Debian maintainer (I do not feel like beeing in this position, thought) is to forward these patches.  So if you consider checking the packaging repository and incorporate what is there your contribution - that's perfect.

Kind regards, Andreas.

larsoner, 2019-01-18:
Great, thanks. I'll close this since it looks like you have a fix that allows us to move forward. Thanks for your help @tillea !

closed 2019-01-18 by larsoner

agramfort, 2019-01-18:
thanks guys

yarikoptic, 2019-08-08:
oh hoh, with ` || true` it is pretty much as useful as not running tests at all since nobody would eyeball logs for failures.  I will try to build now and see where we are on this ATM