mne-tools/mne-python #5816: epochs.plot_psd loses axis labels

massich, 2019-01-06:
I would expect this code to work

```py
from mne.viz.tests.test_epochs import _get_epochs
import matplotlib.pyplot as plt
import mne

epochs = _get_epochs()

# subplot axis properly labeled
_, (ax1, ax2) = plt.subplots(nrows=2)
epochs.plot_psd(picks=mne.pick_types(epochs.info, meg='meg'), ax=[ax1, ax2])

# proper axis labels
epochs.plot_psd(picks=mne.pick_types(epochs.info, meg='mag'))

# missing labels
_, (ax1, ax2) = plt.subplots(nrows=2)
epochs.plot_psd(picks=mne.pick_types(epochs.info, meg='mag'), ax=ax1)
```
 
[image]

agramfort, 2019-01-07:
I can replicate the pb

>

massich, 2019-01-07:
The issue is here:

We recover the figure and if the number of axis match, then we set `make_label=True`.

https://github.com/mne-tools/mne-python/blob/feb0bec3acf334ee391711109c9a541697813a30/mne/viz/raw.py#L600-L602

What I'm not sure if this is a bug or a feature.
This was added by @larsoner [here](https://github.com/mne-tools/mne-python/commit/53506d1d5547a5dbed8e30d9927232a58ced7a81#diff-190df768fb5ee305b0a421d3dd6ae9adR600)

agramfort, 2019-01-07:
I would say it's a bug. It's easier to change the labels afterwards.

massich, 2019-01-08:
>I'm fine with changing this, but make sure the change is in the BUG section because it's a backward incompatible change that will break some people's plotting code. (I'm sure it will break some of mine.)

That's why I asked. What I could not figure out is when this should be false. My understanding is that you add that so that the first plot is possible. (otherwise, it does not). But If we always return True that would work in both cases (first and last). But I could not figure out when we would like to not write the labels.

larsoner, 2019-01-08:
Basically if we create the axes internally, we know (at least) the `xlabel` is placed correctly; if the user supplies them, we can't guarantee it

agramfort, 2019-01-09:
yes that's why it should be displayed by default when you pass one ax

>

PR #7166 (2020-01-06): MRG: Fix missing xlabel for epochs or raw.plot_psd when passing a subset o…
…f axis from a figure

closes #5816
closed 2020-01-06 by agramfort