mne-tools/mne-python #5830: ENH: Example on how to make a figure publication-ready

larsoner, 2019-01-11:
Every time I go to make a M/EEG figure publication ready, I have to go back to old scripts and dig through StackOverflow to make things presentable. And I've been told by other folks that it's a non-trivial task for them. Basically I want to show people how they can avoid using InkScape / Illustrator.

I propose we add an example that contains a bunch of `matplotlib` tricks to take some MEG data figure and show how you can make it publication-quality. I'm thinking the data content would be (from `sample`):

- [ ] ERP plot
- [ ] Brain activation

And tricks included would be:

- Mechanics
   - [ ] using `imshow` after a `brain.screenshot()` to show the `mayavi` activation map in a figure
   - [ ] adding a `matplotlib` colorbar to match the `mayavi` figure (I have separate code for this that we can integrate)
- Aesthetics
   - [ ] removing top and right `spines`
   - [ ] adding a legend, tweaking legend properties
   - [ ] adding *A* and *B* subplot labels to the upper left of figures

Others feel free to chime in with little tricks that you have found useful and would like to see added. Eventually I'm hoping what we produce will be pretty enough to be on the front page carousel!

@drammock this might be a good one for when you need a break from digging through tutorials :)

agramfort, 2019-01-11:
+100

@drammock <https://github.com/drammock> it would be great if you could take
a stab at this

>

drammock, 2019-01-11:
sure, I'll take a swing in the next week or two, when I need a fun plotting break

jasmainak, 2019-01-11:
good idea. It would be nice to show the [before and after](https://journals.plos.org/ploscompbiol/article/figure?id=10.1371/journal.pcbi.1003833.g004) in an example. I do more basic manipulations:
- ticks pointing out and remove two spines
- reduce number of ticks (e.g., remove every other tick and their labels) -- I have to look up stack overflow every single time to do this
- make sure tick labels don't have long decimals like 1.400000001, so format it to have 2 or 3 decimals max
- adjust the height to width ratio (figsize) and the spacing between plots and tight layout while saving them

PR #6745 (2019-09-12): MRG, ENH: Publication tweaking example
Closes #5830.

@drammock feel free to take over, just wanted to put some of my tweaks down somewhere.
closed 2019-09-12 by agramfort