mne-tools/mne-python #5839: Annotations lost when concatenating Raws / incorrect shift calculation in _combine_annotations

hekolk, 2019-01-16:
Annotations are lost when concatenating (BrainVision) raws.

The bug seems to be in ``mne.annotations._combine_annotations`` ( https://github.com/mne-tools/mne-python/blob/master/mne/annotations.py#L362 ).
When concatenating annotations, the time shift calculation seems to be incorrect since the annotations of the second raw are shifted both by the length of the first raw (``one_n_sampes``) and the difference in the start times of the recording (``meas_date - two.orig_time``). Hence, the annotations of the second raw are shifted "too far" and removed when cropping the annotations in ``Raw.set_annotations`` (some annotations might be kept if the lengths of the Raws differs). 

I'm not sure if I'm missing something, but since annotations are supposed to be used instead of stim channels, this bug seems rather critical to me.

#### Steps and/or code to reproduce
On a high level, the bug manifests itself as

```
import mne
import numpy as np

# vhdr_files = ...
#%% load first raw
raw1 = mne.io.read_raw_brainvision(vhdr_files[0])
events1, _ = mne.events_from_annotations(raw1)
print(len(events1)) 
# 977

#%% load second raw
raw2 = mne.io.read_raw_brainvision(vhdr_files[1])
events2, _ = mne.events_from_annotations(raw2)
print(len(events2))
 # 975

#%% combine raw
raw1.append(raw2)
events_comb, _ = mne.events_from_annotations(raw1)
print(len(events_comb))
# 979
np.testing.assert_equal(events1, events_comb[:len(events1), :])
```
All but two events of the second recording are missing.

On a lower level, the shifting becomes obvious:

```
orig_time1 = 0
duration1 = 10.
sfreq = 1000
annot1 = mne.Annotations(onset=[0, 1], duration=[0,0], description=['test', 'test'],
                         orig_time=orig_time1)
annot2 = mne.Annotations(onset=[0, 1], duration=[0,0], description=['test', 'test'],
                         orig_time=orig_time1 + duration1)
#%%
# this call corresponds to the one in Raw.append
annot_combined = mne.annotations._combine_annotations(
    annot1, annot2, one_n_samples=duration1 * sfreq,
    one_first_samp=0, two_first_samp=0, sfreq=sfreq,
    meas_date=orig_time1)
print(annot_combined.onset)
# [ 0.  1. 20. 21.]
```

#### Expected results
The onsets of the annotations should be ``[0., 1., 10. ,11.]``.

#### Actual results
The onsets are ``[ 0.  1. 20. 21.]``

hekolk, 2019-01-17:
One could just replace ``orig_time`` of the annotations of the Raw to be appended with the the initial one, but I'm not sure whether that may have any side effects...

https://github.com/mne-tools/mne-python/blob/master/mne/io/base.py#L2130
```
r_annotations = r.annotations.copy()
r_annotations.orig_time = annotations.orig_time
annotations = _combine_annotations(
    annotations, r_annotations, n_samples,
    self.first_samp, r.first_samp,
    self.info['sfreq'], self.info['meas_date'])

massich, 2019-01-21:
In order to combine annotations, they should share the same `orig_time`.
So in your example, if we do `annot1+annot2` raises an error.

what you are suggesting is that when concatenating raw files, the annotations are broken. I guess that if both raw files share meas_date, that would be no problem.

I'll try to replicate

massich, 2019-01-21:
```py
    import mne
    info = mne.create_info(ch_names=10, sfreq=10.)
    raw_A = mne.io.RawArray(data=np.empty((10, 10)), info=info, first_samp=10)
    raw_A.info['meas_date'] = 0
    annot = mne.Annotations([.5], [.2], ['dummy'], orig_time=None)
    raw_A.set_annotations(annotations=annot)

    print(raw_A)
    print(raw_A.annotations)
    for a in raw_A.annotations:
        print(a)

    # raw_B = raw_A.copy().append(raw_A.copy())
    raw_B = raw_A.copy()
    raw_B.append(raw_A.copy())

    print(raw_B)
    print(raw_B.annotations)
    for a in raw_B.annotations:
        print(a)

```
This actually works as I would expect. (apart from the issue that I cannot assign `raw_B` as I would expect).

Can you share your vhdr files?

massich, 2019-01-21:
Sorry I forgot the output. 
```
<RawArray  |  None, n_channels x n_times : 10 x 10 (0.9 sec), ~24 kB, data loaded>
<Annotations  |  1 segment : dummy (1), orig_time : 1970-01-01 00:00:00>
OrderedDict([('onset', 1.5), ('duration', 0.2), ('description', 'dummy'), ('orig_time', 0.0)])
<RawArray  |  None, n_channels x n_times : 10 x 20 (1.9 sec), ~25 kB, data loaded>
<Annotations  |  4 segments : dummy (2), BAD boundary (1), EDGE boundary (1), orig_time : 1970-01-01 00:00:00>
OrderedDict([('onset', 1.5), ('duration', 0.2), ('description', 'dummy'), ('orig_time', 0.0)])
OrderedDict([('onset', 2.5), ('duration', 0.2), ('description', 'dummy'), ('orig_time', 0.0)])
OrderedDict([('onset', 2.0), ('duration', 0.0), ('description', 'BAD boundary'), ('orig_time', 0.0)])
OrderedDict([('onset', 2.0), ('duration', 0.0), ('description', 'EDGE boundary'), ('orig_time', 0.0)])
```

`raw_A` has a single onset at 1.5 (which is at 0.5 of first_samp)

and `raw_B` has 2 bad segments + at the junction, `meas_date` of the second is ignored and the onset corresponds to 2.5 as if the buffers where concatenated.

massich, 2019-01-21:
On your second example you cannot concatenate annotations with different `orig_times`. 

```py
    orig_time1 = 0
    duration1 = 10.
    sfreq = 1000
    annot1 = mne.Annotations(onset=[0, 1], duration=[0, 0], description='test',
                             orig_time=orig_time1)
    annot2 = mne.Annotations(onset=np.array([0, 1])+duration1, duration=[0, 0],
                             description='test', orig_time=orig_time1)
    print((annot1+annot2).onset)
```

This gives you `[0., 1., 10. ,11.]`, but combining annotations with a different `orig_time` is not possible.

hekolk, 2019-01-21:
Thanks for the clarification @massich ! 

My ``Raw``s have different ``meas_date`` and annotations have different ``orig_time``s [1]. Hence, in my previous example ``raw1.annotations + raw2.annotations`` throws  (``ValueError: orig_time should be the same to add/concatenate 2 annotations [...]``).
However, ``raw1.append(raw2)`` succeeds: This calls ``_combine_annotations``, which tries to compensate for the differing ``orig_time``, but in addition to that also shifts the second ``Annotation``'s onsets by the duration of the first raw => incorrect offsets for the newly appended annotations.
I extended your example to illustrate that [2].

If I understand you correctly, ``raw1.append(raw2)`` (with different ``meas_date``) should currently fail---this would be unfortunate since imho it's a frequent use case to concatenate raw recordings and the basic logic is already there.
Probably, the parameter ``one_n_samples`` should only be passed to ``_combine_annotations`` if the second annotation's ``orig_time`` is ``None``!?

[1] I'm loading an experiment where each block is a separate Brainvision recording. 
In my script, I ``append`` the different ``Raw``s of the blocks into a single one and extract events (previously using the synthesized stim channel, and I encountered this problem when switching to extracting events from annotations after the latest deprecations). During loading of the annotations, the start time of the recording is read from the marker file (for ``info['meas_date']`` in https://github.com/mne-tools/mne-python/blob/master/mne/io/brainvision/brainvision.py#L478 and for the annotations in https://github.com/mne-tools/mne-python/blob/master/mne/io/brainvision/brainvision.py#L226).

[2]
```
import mne
import numpy as np
info = mne.create_info(ch_names=10, sfreq=10.)
raw_A = mne.io.RawArray(data=np.empty((10, 10)), info=info, first_samp=10)
raw_A.info['meas_date'] = 0
annot = mne.Annotations([.5], [.2], ['dummy'], orig_time=None)
raw_A.set_annotations(annotations=annot)

print(raw_A)
print(raw_A.annotations)
for a in raw_A.annotations:
    print(a)

#%% previous example, everything works fine
raw_B = raw_A.copy()
raw_C = raw_A.copy()
raw_B.append(raw_C)

print(raw_B)
print(raw_B.annotations)
for a in raw_B.annotations:
    print(a)
# <RawArray  |  None, n_channels x n_times : 10 x 10 (0.9 sec), ~24 kB, data loaded>
# <Annotations  |  1 segment : dummy (1), orig_time : 1970-01-01 00:00:00>
# OrderedDict([('onset', 1.5), ('duration', 0.2), ('description', 'dummy'), ('orig_time', 0.0)])

#------------------------------
# if the second orig_time is "after the end" of the first recording, the second dummy annotation 
# is lost
raw_B = raw_A.copy()
raw_C = raw_A.copy()

raw_C.info['meas_date'] = 3
raw_C.annotations.orig_time = raw_C.info['meas_date']
raw_B.append(raw_C)
# RuntimeWarning: Omitted 1 annotation(s) that were outside data range.

print(raw_B)
print(raw_B.annotations)
for a in raw_B.annotations:
    print(a)
# <RawArray  |  None, n_channels x n_times : 10 x 20 (1.9 sec), ~25 kB, data loaded>
# <Annotations  |  3 segments : dummy (1), BAD boundary (1), EDGE boundary (1), orig_time : 1970-01-01 00:00:00>
# OrderedDict([('onset', 1.5), ('duration', 0.2), ('description', 'dummy'), ('orig_time', 0.0)])
# OrderedDict([('onset', 2.0), ('duration', 0.0), ('description', 'BAD boundary'), ('orig_time', 0.0)])
# OrderedDict([('onset', 2.0), ('duration', 0.0), ('description', 'EDGE boundary'), ('orig_time', 0.0)])

#------------------------------
# if we second orig_time is still within the duration of the first Raw, the annotation is kept,
# but will have an incorrect onset
raw_B = raw_A.copy()
raw_C = raw_A.copy()

raw_C.info['meas_date'] = 0.5
raw_C.annotations.orig_time = raw_C.info['meas_date']
raw_B.append(raw_C)
# RuntimeWarning: Limited 1 annotation(s) that were expanding outside the data range.

print(raw_B)
print(raw_B.annotations)
for a in raw_B.annotations:
    print(a)
# <RawArray  |  None, n_channels x n_times : 10 x 20 (1.9 sec), ~25 kB, data loaded>
# <Annotations  |  4 segments : dummy (2), BAD boundary (1), EDGE boundary (1), orig_time : 1970-01-01 00:00:00>
# OrderedDict([('onset', 1.5), ('duration', 0.2), ('description', 'dummy'), ('orig_time', 0.0)])
# OrderedDict([('onset', 3.0), ('duration', -1.6653345369377348e-16), ('description', 'dummy'), ('orig_time', 0.0)])
# OrderedDict([('onset', 2.0), ('duration', 0.0), ('description', 'BAD boundary'), ('orig_time', 0.0)])
# OrderedDict([('onset', 2.0), ('duration', 0.0), ('description', 'EDGE boundary'), ('orig_time', 0.0)])
```

agramfort, 2019-01-22:
it should not fail to concat raws with different meas_date as indeed it's a
really frequent usecase.
mne inserts a bad annotation to notifiy there is file boundary and it
should just work ie the orig_time
of the appended raw file should be adjust to just work.

>

PR #5851 (2019-01-23): [MRG][Fix] Concatenate raw objects with annotations
Fix #5839, #5555 

This PR fixes the fact that `raw_A.append(raw_B)` ignores `meas_date` and
supposes that the buffers of `raw_A` and `raw_B` where captured one right
after the other.
commit 7331ad6e (2019-01-23): [Fix] Concatenate raw objects with annotations (#5851)

Fix #5839, #5555 

This PR fixes the fact that `raw_A.append(raw_B)` ignores `meas_date` and
supposes that the buffers of `raw_A` and `raw_B` where captured one right
after the other.
closed 2019-01-23 by massich