mne-tools/mne-python #5847: Question about mne.time_frequency.psd_array_multitaper

tang37, 2019-01-21:
Hello,

I wanted to use mne.time_frequency.psd_array_multitaper to get a power spectrum of local field potential data, but I got negative power values (around 40-100Hz somehow). I put 'psd, freqs = psd_array_multitaper(data, 30000, adaptive=True, normalization='full', verbose=0)'. What could possibly go wrong? 

Any help would be appreciated!
Yu

JoseAlanis, 2019-01-22:
Hey Yu, are you shure it's negative numbers and not something like `2.36e-11` (which is just very very small)? I don't know if the PSD of one signal can be nagative. I don't think so, at least not without transfomation to decibels or something like that.

You could try plotting the result of `mne.time_frequency.psd_array_multitaper()` and see if you really get something below 0. You could try something like this (see [here](https://martinos.org/mne/stable/auto_tutorials/plot_sensors_time_frequency.html#sphx-glr-auto-tutorials-plot-sensors-time-frequency-py)):

```python
import numpy as np
import matplotlib.pyplot as plt
import mne

f, ax = plt.subplots()
psds, freqs = mne.time_frequency.psd_array_multitaper(data, 
                                                      sfreq=30000, adaptive=True, normalization='full', verbose=0)

# get mean and sd
psds_mean = psds.mean(0).mean(0)
psds_std = psds.mean(0).std(0)
# plot results
ax.plot(freqs, psds_mean, color='k')
ax.fill_between(freqs, psds_mean - psds_std, psds_mean + psds_std,
                color='k', alpha=.5)
plt.show()
```

tang37, 2019-01-22:
Hi Jose,

Thank you for the reply! I double checked my script, and found out that the power density values were converted to decibels halfway in the code. When I computed band powers using areas under the curve, it gave me negative values. It was a very stupid mistake, thank you for pointing it out!

Yu

closed 2019-01-22 by agramfort