mne-tools/mne-python #5850: Unable to include all vertices in a single Label (Source Estimation)

dasdiptyajit, 2019-01-22:
Hi guys,

I would like to restrict dipoles or source activity within a particular Brain Label. So, I read the subject specific 'label' from freesurfer and pass these vertices during source estimation. Now, when I read the single label, it contains more dipole locations, however, I always ended up with less no. of dipole locations after the source estimation. It seems like some dipoles/vertices indices within the label is not present in the source space, therefore they are missing in STC. 

Ps: I am working on a single subject data, so I am not using label from 'fsaverage'. How to include this missing dipole locations/vertices in source space data?

Best,
Dip

larsoner, 2019-01-22:
>  It seems like some dipoles/vertices indices within the label is not present in the source space, therefore they are missing in STC.

Yes this is the way it is designed to work. In general you subsample from the dense (~150k verts) mesh to get the source space for each hemi (~4k verts). For M/EEG data you will probably not gain anything from including a denser mesh, but things will take much longer and memory requirements will get much worse. In other words, in most applications you should be fine using this subselection of vertices per label rather than all the vertices in the label in the dense mesh.

dasdiptyajit, 2019-01-22:
@larsoner thanks for the reply. Are subsection of vertices randomly dropped from a source space? I am working with some auditory data and particularly interested in negative potential of the current flow.  When I try to extract the activity from left hemisphere with ''transversetemporal' label, I obtain mainly upward current/dipoles activity (with sign), which I am not sure, the case should be. Is there any way to cross check the dropped vertices activity within the same label?

larsoner, 2019-01-22:
> Are subsection of vertices randomly dropped from a source space?

No they are subselected from the dense inflated sphere based on a recursively subdivided icosehedron or octahedron, i.e. evenly on the sphere and thus in a principled way on the non-inflated surface.

>  Is there any way to cross check the dropped vertices activity within the same label?

No. But if you are using a minimum-norm solver, in general nearby vertices are going to have very similar activity anyway.

See this example for further ideas:

https://martinos.org/mne/stable/auto_examples/inverse/plot_label_source_activations.html

Since this is drifting away from bugs/features and more to interpretation, feel free to follow up with other questions on the listserv rather than here.

closed 2019-01-22 by larsoner