mne-tools/mne-python #5854: ENH: Annotations.sort

larsoner, 2019-01-22:
Sort annotations based on onset. Should return `serf` for consistency.

massich, 2019-01-23:
If we have a sorting method, we should used in places like this (there are more):
https://github.com/mne-tools/mne-python/blob/e96d706b65b6c6a21821d3dcbc06e10903e9c05b/mne/viz/utils.py#L1913-L1914

The other option it would be to actually keep annotations always sorted by onset, and use `numpy.searchsorted` in `append` and `_combine_annotations`

mmagnuski, 2019-01-24:
I'd be +1 for always sorting, but I'm also ok with `annotations.sort()`.

agramfort, 2019-01-25:
+1 for always sorting

>

cbrnr, 2019-01-25:
+1 for always sorting and adding an extend method. As tie breakers I suggest that we use description and then duration.

larsoner, 2019-01-29:
It looks like `append` already has `extend`-like behavior, so we can just update the docstring.

PR #5900 (2019-02-11): MRG+1: Keep annotations sorted
Closes #5854.

Would be good to merge #5895 first, which contains some `Annotations`-related tweaks and new tests.
closed 2019-02-11 by larsoner