mne-tools/mne-python #5856: save-load FIF round trip fails due to Annotations

massich, 2019-01-23:
```py
import os.path as op
import numpy as np
from numpy.testing import (assert_array_almost_equal, assert_array_equal,
                           assert_allclose)

from mne.datasets import testing
from mne.io import read_raw_fif
from mne.utils import _TempDir
from mne.annotations import Annotations

testing_path = testing.data_path(download=False)
data_dir = op.join(testing_path, 'MEG', 'sample')
fif_fname = op.join(data_dir, 'sample_audvis_trunc_raw.fif')
tmpdir = _TempDir()

raw_1 = read_raw_fif(fif_fname, preload=True)
raw_1.set_annotations(Annotations(np.arange(20), np.ones((20,)), 'test'))
split_fname = op.join(tmpdir, 'split_raw.fif')
raw_1.save(split_fname, buffer_size_sec=1.0, split_size='10MB')
raw_2 = read_raw_fif(split_fname)

print(raw_1.annotations)
print(raw_2.annotations)

assert_array_almost_equal(raw_1.annotations.onset, raw_2.annotations.onset)
```

I get two extra annotations at the end of the file that I've no idea where are they coming from.

```
<Annotations  |  20 segments : test (20), orig_time : 2002-12-03 19:01:10.720100>
<Annotations  |  22 segments : test (22), orig_time : 2002-12-03 19:01:10.720100>
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
~/code/mne-python/test_xx.py in <module>
     24 print(raw_2.annotations)
     25 
---> 26 assert_array_almost_equal(raw_1.annotations.onset, raw_2.annotations.onset)
     27 

~/miniconda3/envs/mne/lib/python3.6/site-packages/numpy/testing/_private/utils.py in assert_array_almost_equal(x, y, decimal, err_msg, verbose)
    971     assert_array_compare(compare, x, y, err_msg=err_msg, verbose=verbose,
    972              header=('Arrays are not almost equal to %d decimals' % decimal),
--> 973              precision=decimal)
    974 
    975 

~/miniconda3/envs/mne/lib/python3.6/site-packages/numpy/testing/_private/utils.py in assert_array_compare(comparison, x, y, err_msg, verbose, header, precision, equal_nan, equal_inf)
    738                                 verbose=verbose, header=header,
    739                                 names=('x', 'y'), precision=precision)
--> 740             raise AssertionError(msg)
    741 
    742         flagged = bool_(False)

AssertionError: 
Arrays are not almost equal to 6 decimals

(shapes (20,), (22,) mismatch)
 x: array([42.955971, 43.955971, 44.955971, 45.955971, 46.955971, 47.955971,
       48.955971, 49.955971, 50.955971, 51.955971, 52.955971, 53.955971,
       54.955971, 55.955971, 56.955971, 57.955971, 58.955971, 59.955971,
       60.955971, 61.955971])
 y: array([42.955971, 43.955971, 44.955971, 45.955971, 46.955971, 47.955971,
       48.955971, 49.955971, 50.955971, 51.955971, 52.955971, 53.955971,
       54.955971, 55.955971, 56.955971, 57.955971, 58.955971, 59.955971,
       60.955971, 61.955971, 62.958802, 62.958802])
```

agramfort, 2019-01-26:
when you split the files the annotations get cut to not exceed file time domain. So when you concat on read the annotations that overlap with the split location get divided in 2 pieces. We would need a mechanism to stitch them back together.

massich, 2019-05-15:
#5857 fixes this issue, but the behavior does not correspond to [this](https://github.com/mne-tools/mne-python/issues/5856#issuecomment-457814081):

>when you split the files the annotations get cut to not exceed file time domain. So when you concat on read the annotations that overlap with the split location get divided in 2 pieces. We would need a mechanism to stitch them back together.

The actual behavior saves all the `annotations` in all `fif` parts. When loading a split file the annotations are taken from the first raw file. I have no trouble with this implementation. It has the advantage that when distributing a *single part* the annotations work out of the box. See the following snipped:

```py
from os import path as op
from shutil import copyfile
from os import listdir

import numpy as np

from mne.datasets import testing
from mne.io import read_raw_fif
from mne.utils import _TempDir
from mne import Annotations

testing_path = testing.data_path(download=False)
data_dir = op.join(testing_path, 'MEG', 'sample')
fif_fname = op.join(data_dir, 'sample_audvis_trunc_raw.fif')
tempdir_A = _TempDir()
tempdir_A, tempdir_B = [_TempDir() for _ in range(2)]

# load a file and save it as split
raw_1 = read_raw_fif(fif_fname, preload=True)
annot = Annotations(np.arange(20), np.ones((20,)), 'test')
raw_1.set_annotations(annot)
split_fname = op.join(tempdir_A, 'split_raw.fif')
raw_1.save(split_fname, buffer_size_sec=1.0, split_size='6MB')
print('files saved: ', listdir(tempdir_A))

# copy a single part somewhere else
raw_part_fname = op.join(tempdir_B, 'part.fif')
copyfile(src=op.join(tempdir_A, 'split_raw-2.fif'), dst=raw_part_fname)
print('files saved: ', listdir(tempdir_B))

# load a single part and check what happens
raw_2 = read_raw_fif(raw_part_fname)
for _annot in raw_2.annotations:
    print(f"onset: {_annot['onset']}, duration: {_annot['duration']}")
```

```sh
onset: 50.97441875934601, duration:0.981552004814148
onset: 51.955970764160156, duration:1.0
onset: 52.955970764160156, duration:1.0
onset: 53.955970764160156, duration:1.0
onset: 54.955970764160156, duration:0.0276719331741333
```

If we want to change the behavior of how things are done we can do that after the release. @larsoner WDYT?

agramfort, 2019-05-15:
+1 to save annotations only in the first chunk/split 
by doing this no more pain of stitching back annotations from different files.

larsoner, 2019-05-15:
Agreed, better not to have to split+stitch if we can avoid it

massich, 2019-05-15:
(maybe I've misexpressed) I would change nothing unless we have annotations that weight gigabytes. The set annotations by the called when reading the part handle the annotations properly. And after #5857 there is no stitching.

PR #5857 (2019-05-15): [MRG] FIX: save-load split FIF round trip
split save and reload a FIF file fails due to Annotations

Fixes #5856
closed 2019-05-15 by larsoner