mne-tools/mne-python #5860: Processing all channels in one go with `Raw.apply_function`

hubertjb, 2019-01-24:
I want to apply a function on the internal data in a `Raw` object, and I saw that there is an `apply_function()` method for this purpose. However, `apply_function` seems to apply the function channel-wise (sequentially or in parallel), while my function could potentially be run faster by working on all channels at once.

Would there be any interest in adding an attribute to `apply_function` to allow applying the function on all channels at once instead of channel-by-channel?

I had something like this in mind:

```python
if n_jobs == 1:
    # modify data inplace to save memory
    if apply_in_one_go:
        self._data[picks, :] = _check_fun(fun, data_in[picks, :],
                                           *args, **kwargs)
    else:
        for idx in picks:
            self._data[idx, :] = _check_fun(fun, data_in[idx, :],
                                            *args, **kwargs)
```

If this is of interest I can make a PR.

Thanks!

massich, 2019-01-24:
I think it makes more sense to do the opposite, the function should always work with the data in one go, and delegate the operation to the call.

larsoner, 2019-01-24:
> But this might not be backward compatible. So we should add the parameter and deprecate it, in one go.

I wouldn't bother with a deprecation cycle, it seems overkill. But sure, if `channel_wise=True` (default) is `False`, then ignore `n_jobs` and just pass the entire data array.

massich, 2019-01-24:
+1 for `channel_wise=False`

hubertjb, 2019-01-24:
Perfect, I'll make a PR to add 'channel_wise' then. Thanks!

larsoner, 2019-01-24:
Please leave the default as `True` for backward compat

larsoner, 2019-01-24:
Argh I mean `False`, mis-read things...

larsoner, 2019-01-24:
Double mis-read, `channel_wise=True` is what we do now and it should remain the default

massich, 2019-02-15:
@hubertjb can we close this?

larsoner, 2019-02-15:
Yes the PR has been merged

closed 2019-02-15 by larsoner