mne-tools/mne-python #5863: GDF events

massich, 2019-01-24:
Check this MWE. `mne.events_from_annotations` is not able to pick up the different events and they all get mapped to `'GDF event'`. Plus I'm not sure if we should expose `mne.io.find_edf_events` nor `edf_raw.find_edf_events()` because that would now work with `raw.find_edf_events()`. (and I would expect the same methods, no matter which reader did I use to load `raw`)

MWE
```py
"""
Disclaimer this MWE uses data from http://www.bbci.de/competition/iv/#dataset2a
"""
import mne
import numpy as np
from os import path as op

tmpdir = mne.utils._TempDir()
DROPBOX_FILE_URL = "https://www.dropbox.com/s/nn6dzc5cc7riz7c/A01E.gdf?dl=1"
fname = op.join(tmpdir, 'A01E.gdf')

mne.utils._fetch_file(url=DROPBOX_FILE_URL, file_name=fname,
                      hash_='ec3de227cd3bf1760accca0e3269590e')

raw = mne.io.read_raw_edf(fname)

# Regular workflow
events_A, event_id_A = mne.events_from_annotations(raw)
print(raw.annotations)
print(events_A.shape)
print(event_id_A)

# whatever
def my_events_from_annotations(edf_raw):
    _, onset, label, duration, _ = edf_raw.find_edf_events()
    events = np.stack([onset, duration, label], axis=1)
    event_id = {'GDF event {}'.format(chr(65 + idx)): l
                for idx, l in enumerate(set(label))}
    return events, event_id

events_B, event_id_B = my_events_from_annotations(raw)

print(events_B.shape)
print(event_id_B)

np.testing.assert_array_equal(events_A[:, 0], events_B[:, 0])
```
output
```<Annotations  |  595 segments : GDF event (595), orig_time : 2005-01-19 12:00:00>
(595, 3)
{'GDF event': 1}
(595, 3)
{'GDF event A': 768, 'GDF event B': 783, 'GDF event C': 1072, 'GDF event D': 276, 'GDF event E': 277, 'GDF event F': 32766, 'GDF event G': 1023}
```

ping: @jaeilepp

massich, 2019-01-24:
cc: @cbrnr do you know the data? your name appears in the website as organizer

massich, 2019-01-24:
@larsoner `find_edf_events` was introduced in `0.15` and we did not think of removing it in `0.17`.

Now my question is what would we expect from `events_from_annotations`? check the usage of events ids here
https://github.com/robintibor/braindecode/blob/master/braindecode/datasets/bcic_iv_2a.py#L49

I'm not sure where those ids come from.

massich, 2019-01-24:
```
import mne
fname = mne.datasets.sample.data_path() + '/MEG/sample/sample_audvis_raw.fif'
raw = mne.io.read_raw_fif(fname)
mne.io.find_edf_events(raw)
```

cbrnr, 2019-01-24:
Yes, I know the data. I'll take a look tomorrow.

cbrnr, 2019-01-25:
Wait, am I missing something? When I load the data, there are no annotations. GDF stores annotations as discrete events, so I was expecting that they end up in `raw.annotations` (I tested with 0.17).

cbrnr, 2019-01-25:
Instead, a stim channel is synthesized, which I think should not have happened.

massich, 2019-01-25:
What do you get with Master?

massich, 2019-01-25:
The files we are using for testing GDF do not contain proper labels. What do you think about distributing the file I shared here as test?

massich, 2019-01-25:
(and just as a question) why the code of GDF is found in EDF if they do not share anything?

cbrnr, 2019-01-25:
> The files we are using for testing GDF do not contain proper labels. What do you think about distributing the file I shared here as test?

Which GDF files do we currently have for testing? I don't think you can distribute the file A01E.gdf, because it doesn't have a license (although the same file is available under a CC license from http://bnci-horizon-2020.eu/database/data-sets). Furthermore, it is a bit large to be included in our tests, so I'd prefer to have a very short GDF with events.

> (and just as a question) why the code of GDF is found in EDF if they do not share anything?

Good question. From a user perspective, I'd prefer a separate function `read_raw_gdf`, but internally GDF uses the same basic EDF specs, but defines its own header plus an event table (but I'd argue that these changes are enough to warrant its own reader function).

I've dug through the code, and the types are imported correctly in `_read_gdf_header`. They are available in the `edf_info` dict, the `'events'` key contains this information as the list element 3.

Where do you create annotations? I think we could directly use `edf_info["events"]` to correctly map to annotation descriptions.

agramfort, 2019-01-25:
I am -1 on a new reading function. I would first fix the reader. Is there a
fundamental problem?

cbrnr, 2019-01-25:
No, there's no fundamental problem. I just think that GDF files should be opened with `read_raw_gdf`, which would be easier to find for people. Of course after the reader is fixed.

agramfort, 2019-01-25:
ok but I would not make this a priortiy. It's easy to find on google
that mne can read gdf with read_raw_edf

I think we have more urgent needs of we have some spare cycles.

cbrnr, 2019-01-25:
Maybe something as simple as #5877.

agramfort, 2019-01-26:
this is quite a hack... we should one way of going things.
If you want to go this route, we need to deprecate gdf support from
read_raw_edf
and adjust documentation accordingly

>

PR #5866 (2019-01-27): [MRG][FIX] Fix GDF returning all annotations with same description
The snipped in #5863 now returns this:

``` shell
<Annotations  |  595 segments : 32766 (9), 276 (1), 277 (1)..., orig_time : 2005-01-19 12:00:00>
(595, 3)
{'32766': 1, '276': 2, '277': 3, '1072': 4, '768': 5, '783': 6, '1023': 7}
(595, 3)
{'GDF event A': 768, 'GDF event B': 783, 'GDF event C': 1072, 'GDF event D': 276, 'GDF event E': 277, 'GDF event F': 32766, 'GDF event G': 1023}
```

I'm not sure if the ids, are stored somewhere in the meatadata (ie: `4242` meaning something like `Start recording`)

fixes #5863
closed 2019-01-27 by agramfort